Ultimaker/Cura #607: 15.06.03 does not start

scheny, 2016-01-21:
Version: 15.06.03
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 68, in readFromFile
  File "c:\dev\python34\lib\_collections_abc.py", line 504, in __iter__
  File "c:\dev\python34\lib\configparser.py", line 1204, in __getitem__
  File "c:\dev\python34\lib\configparser.py", line 773, in get
  File "c:\dev\python34\lib\configparser.py", line 374, in before_get
  File "c:\dev\python34\lib\configparser.py", line 423, in _interpolate_some
configparser.InterpolationSyntaxError: '%' must be followed by '%' or '(', found: "%20Elk/Drivetrain\\xff\\xff\\xff\\xff\\x00\\x00\\x00\\x01'"

It seems to have problems with some file names

Ghostkeeper, 2016-01-22:
I seem to remember something like this coming by. Would you like to try if it still occurs with a more recent version? It might already have been fixed since the version you're using is from June.

You can find newer versions here: http://software.ultimaker.com/Cura_closed_beta/15.10/

scheny, 2016-01-22:
Works with version 15.10, you can close this issue.

Problem with 15.10 is it has so many bugs I can't even tell where to begin. Main issue is, that slicing often does not start, but this seems to be a commonly known problem.

closed 2016-01-22 by scheny

nallath, 2016-01-22:
We almost have a new release which solves a -lot- of bugs. Should be released soon.