Ultimaker/Cura #609: [2.1] i18n: Hardcoded default strings

thopiekar, 2016-01-22:
While translating some strings I noticed hardcoded strings like the following line:
https://github.com/Ultimaker/Cura/blob/master/resources/qml/JobSpecs.qml#L161

https://github.com/Ultimaker/Cura/blob/2.1/resources/i18n/de/cura.po#L634
https://github.com/Ultimaker/Cura/blob/2.1/resources/i18n/de/cura.po#L639

So the problem is that these lines appear in the pot and po files twice, once as "0.0 m" and then as "%1 m". So you need to give two translations for this.

I guess it would be better to use "%1 m" and set "0.0" for "%1" while loading the UI.

nallath, 2016-01-22:
Nice catch! Unfortunately, we're in a string freeze now, so it will remain untranslated for the moment.

thopiekar, 2016-01-22:
Ok, I thought now more about the problem and even think this doesn't need to be translated at all.

Another line like this is https://github.com/Ultimaker/Cura/blob/2.1/resources/i18n/de/cura.po#L56 :
"%(width).1f x %(depth).1f x %(height).1f mm"

"mm" is an SI unit and even changing it here does not convert the length to this unit. The only part which might need a translation is "%(width).1f x %(depth).1f x %(height).1f", as other regions and cultures expect another order of these values.

nallath, 2016-01-22:
Enabling the order to be changed means that we have to add it to the PO files. As they are now being translated, this is not possible atm.

thopiekar, 2016-01-22:
That what I meant, but do we really need to translate the "mm"?

Btw., is someone already working on a German translation? Just prepared a
PR with some translations today.

nallath, 2016-01-22:
MM we don't need to translate. As far as I we pay people to translate to a number of languages with german being one of them.

thopiekar, 2016-01-22:
Ok then. In case they only translate the po files you can cherry-pick https://github.com/thopiekar/Cura/commit/b0061af6cc558621eb018ea81ca98df03d3ac97e from my PR. The application icon on Linux has not been translated so far.

Ghostkeeper, 2016-01-25:
The desktop file is not translated since it can't programmatically read the .po files. Thanks for translating it! It's something we should provide to our translators some time to translate as well.

closed 2016-04-14 by thopiekar