Ultimaker/Cura #622: [2.1] Failed to parse socket error: 'UnknownMessageError' object has no attribute 'errno'

thopiekar, 2016-01-29:
Today, when opening an object in Cura I get this message:
```
2016-01-29 17:14:55,390 - DEBUG - Killing engine process
2016-01-29 17:14:55,989 - INFO - Backend crashed or closed. Restarting...
2016-01-29 17:14:56,273 - INFO - Started engine process: /usr/bin/CuraEngine
2016-01-29 17:14:56,277 - DEBUG - Backend connected on port 49674
2016-01-29 17:14:56,800 - DEBUG - Sending data to engine for slicing.
2016-01-29 17:14:58,772 - ERROR - Failed to parse socket error: 'UnknownMessageError' object has no attribute 'errno'
2016-01-29 17:14:59,314 - INFO - Started engine process: /usr/bin/CuraEngine
2016-01-29 17:14:59,323 - DEBUG - Backend connected on port 49675
2016-01-29 17:15:54,699 - DEBUG - Autosaving preferences, instances and profiles
```

I think it is related to https://github.com/Ultimaker/Cura/commit/c53969f1bda90472a2edef003ff3a74ef45938c6

ricardopvz, 2016-01-31:
I have the same problem.
The reason is that CuraEngine doesn't have the `SlicingFinished` message registered and so it sends it with the id `0` instead of `8`. Then `libArcus` throws a `UnknownMessageError` exception because CuraEngineBackend didn't register the message `SlicingFinished` with id `0` but with `8`.

This needs to be fixed here: https://github.com/Ultimaker/CuraEngine/blob/2.1/src/commandSocket.cpp#L103
Add this `private_data->socket->registerMessageType(8, &cura::proto::SlicingFinished::default_instance());`

thopiekar, 2016-01-31:
@ricardopvz So I guess it is really tested? ^^'

In this case I'll make a PR, so the Ultimaker people can easily merge it tomorrow.

ricardopvz, 2016-01-31:
@thopiekar  I didn't really test it. But I changed the id to `0` here https://github.com/Ultimaker/Cura/blob/2.1/plugins/CuraEngineBackend/CuraEngineBackend.py#L242 and it worked.
I use your PPA, so for me it's easier to change the python file than cloning the repos and compiling all from source.

thopiekar, 2016-01-31:
@ricardopvz Great to here someone is using my PPA :smile: 
You probably noticed I was playing around the last commits this weekend.

I'll add the l18n- and this fix to the build recipe. Let's see what we'll get.

ricardopvz, 2016-01-31:
Oh btw, you need to update your ppa with the most recent commit of Cura, or else it will not have this commit https://github.com/Ultimaker/Cura/commit/c53969f1bda90472a2edef003ff3a74ef45938c6, which has the `self._socket.registerMessageType(8, Cura_pb2.SlicingFinished)` and will still throw the same exception.

thopiekar, 2016-01-31:
Just doing it right now. Will update now Cura with my PR here and the PR in CuraEngine about this problem here.

ricardopvz, 2016-01-31:
Yay! It works!
But now there's another problem. The print time doesn't update. 
This is because `CuraEngine` restarts [here](https://github.com/Ultimaker/Cura/blob/2.1/plugins/CuraEngineBackend/CuraEngineBackend.py#L216) and it it never sends the `ObjectPrintTime` message.

thopiekar, 2016-02-01:
Yes, I also noticed that.

nallath, 2016-02-01:
UnknownMessageError -> This is usually caused when the .proto file that is used for the engine is different then the one used for the GUI. We're messing with how libArcus handles messages, but I expect this to be fixed soon.

Ghostkeeper, 2016-02-01:
This was very stupid of me. I made the change. It worked on my computer so I pushed it at the Cura and Uranium repositories. But I forgot to push it on the CuraEngine repository. It's good that you found the issue yourselves, but I did a similar fix last Friday.

Ghostkeeper, 2016-02-01:
The commits f763edfb05ae6a95b9679c92b84846842c580e16, 0e662d7d675eb9074424c1a8a281a6119113d1f3 and 94d8c3ff320a7ac532990d965ee94f8be5c2dec2 were made last Friday but pushed today. It includes the fixes of both https://github.com/Ultimaker/CuraEngine/pull/298 and https://github.com/Ultimaker/CuraEngine/pull/299, so I'm going to close those pull requests. I'm sorry, guys!

thopiekar, 2016-02-01:
No problem :+1:

ricardopvz, 2016-02-01:
It's ok. I'm glad it's fixed!
I guess this issue can be closed now...

closed 2016-02-02 by Ghostkeeper