Ultimaker/Cura #627: [2.1] No firmware found for printer

thopiekar, 2016-02-04:
In the past I remember there was a dialog poping up when trying to update the firmware.
Now the "Updating firmware..." dialog is doing nothing and not asking for the *.hex anymore.

[image]

```
2016-02-04 19:12:50,315 - WARNING - No firmware found for printer /dev/ttyUSB0
```

Ghostkeeper, 2016-02-05:
Yeah, we noticed that. This issue has actually existed since November! It's scheduled to be fixed for the 2.1 release and marked as a blocker, but it's a complex issue to resolve. I might get around to it this afternoon even, but it'll be a tough one to fix.

thopiekar, 2016-02-05:
Ok, hope you'll be successful getting it fixed soon. If you like I can move
the Uranium translations (#628) to the Uranium tree, just to save your time.

Ghostkeeper, 2016-02-05:
Fixed in the 2.1 branches of Cura, Uranium and cura-binary-data.

Ghostkeeper, 2016-02-05:
Oops, wrong issue. The translations were fixed, not this one.

thopiekar, 2016-03-10:
Update: The dialog is not blocking anymore, but choosing a *.hex file is still not possible.

Ghostkeeper, 2016-03-11:
Ah yes, the firmware uploading should be fixed by now. I did a good bit of work on that here:

https://github.com/Ultimaker/cura-binary-data/commit/7dcd9f01110609caf2ddf3811f58bfece39560c0
https://github.com/Ultimaker/Cura/commit/c41b9fb09afa5e0c86474aa6138881bb42027597
https://github.com/Ultimaker/Cura/commit/9379b0a97ba85bcf9aa4bb73f8d6be790799995c

It was tested quite thoroughly with all UM printers.

There is no feature to choose your own *.hex files. Is that something you'd like to have? If so, I'd suggest to make a new issue for it. You can currently upload your own *.hex files by overwriting the existing *.hex files in the program directory, but that is obviously just a workaround.

closed 2016-03-11 by Ghostkeeper

thopiekar, 2016-03-11:
Ah, ok. I didn't notice the changes in cura-binary-data.
Why are the .hex's only there?

Ghostkeeper, 2016-03-14:
The *.hex files are there because we didn't want to bring in the Arduino compiler as extra dependency. So the firmware is pre-compiled offline. I can't say I really agree since it makes Cura less reproducible but writing the build script is very hard™ so I should not complain ;)

The binary files are in a separate repository for style and performance (makes cloning the Cura repository faster since you don't have to download the full history of those binary files).

thopiekar, 2016-03-14:
Ok, makes absolutely sense. I'll try to add the hex binaries into the packages of my PPA, too.
Btw. I made a cleanup in a new PR. Hope you'll like it :wink: