Ultimaker/Cura #634: [2.1] Can't lie a hook flat on the plate

thopiekar, 2016-02-07:
I just designed a simple hook and Cura has difficulties to place it flat on the plate.
It reminds me of using the "method of the golden section" (hope it is the right translation) and the solution is jumping between two solutions all the time.

[image]

http://www.thingiverse.com/thing:1323111/#files

fieldOfView, 2016-02-08:
The LayFlat algorithm has not changed from legacy-Cura. It is a bit of an oversimplification of the problem, because it does not take the center of mass or the stability into account. it just tries to find 3 points in a plane, starting from the lowest vertex in the current orientation. If you first manually rotate your object, you will get a different result.

BagelOrb, 2017-07-21:
It is laid flat on the small face on the bottom.
This is intended behavior.

closed 2017-07-21 by BagelOrb