Ultimaker/Cura #635: [2.1] Legacy dialog used for "Save as file"

thopiekar, 2016-02-07:
This is what I currently see on KDE:
[image]

And what I expect like we have as "open file" dialog:
[image]

awhiemstra, 2016-02-18:
This is intentional due to a bug in the KDE file dialog that I still need to properly report. The KDE dialog converts the Qt file type filters into something the KDE dialog expects. However, unfortunately it only looks at file extensions. Since we have two file types with the same extension, the file dialog does not restore to the right extension. Now, to be fair, the issue arises because of having save support for both ascii and binary STL and there is little reason to save to ascii, but the issue remains that if anything we save to has the same extension as something else, the KDE dialog gets confused.

awhiemstra, 2016-02-18:
And upstream bug filed: https://bugs.kde.org/show_bug.cgi?id=359528

thopiekar, 2016-02-18:
Thanks :+1:

nallath, 2017-07-27:
This can be closed. This is either fixed or out of our hands.

closed 2017-07-27 by nallath