Ultimaker/Cura #637: please tag a new release

greyltc, 2016-02-15:
Your latest release (15.06.03) has been broken for some time (`AttributeError: module 'subprocess' has no attribute 'mswindows'`). Thankfully, a fix for this has been in your code base also for some time. Please, if you get a chance, tag a new release so that I know what's stable.

This would make the jobs of package maintainers (like me, I package cura for Arch) much easier.

fieldOfView, 2016-02-15:
Work is currently going on in the 2.1 branches of the repositories, and we're inching closer and closer to a stable release there.

greyltc, 2016-02-15:
Cool! Looking forward to the release. I guess 15.10 will never see the light of day then?

BTW, sorry I think I actually put this issue in the wrong project, it really belongs in the Uranium repo, (the culprit is [right here](https://github.com/Ultimaker/Uranium/blob/15.06.03/UM/Backend/Backend.py#L92)), but it looks like the two repos seem to be pretty tightly coupled anyway so this release issue is probably still relevant here.

fieldOfView, 2016-02-15:
We're switching to semantic versioning with the new Cura. The YY.MM version numbering got too confusing (especially with the new Cura vs old Cura)

greyltc, 2016-02-15:
Yeah, I figured 2.1 wasn't the January 2002 release ;-)

Ghostkeeper, 2016-02-15:
It's the release from January 1402 by Shire reckoning.

Ghostkeeper, 2016-08-05:
2.1.3 is tagged.

closed 2016-08-05 by Ghostkeeper