Ultimaker/Cura #639: Automatic downscaling in Cura 15.06.xx

Beierxxx, 2016-02-17:
Within Cura 15.06.02 and .03, a STL file will downscaled by around 10% without notification, even scaling is set to 100%.
The same file opened in Cura 15.04.4 is scaled correctly.
[cura bug.docx](https://github.com/Ultimaker/Cura/files/134437/cura.bug.docx)

nallath, 2016-02-17:
This is fixed in the newer versions (15.10 and 2.1). This will not be fixed for 15.06.

closed 2016-02-17 by nallath

Beierxxx, 2016-02-17:
Thanks!

On the Ultimaker website I only can find Cura (for Windows) versions 15.04 and 15.06 (Beta).

When the 15.10 version will be released?

What is the version 2.1 mentioned below ?

Best regards

Axel

fieldOfView, 2016-02-17:
15.10 will not be released. 2.1 will be the next version that will be released. We are close to releasing a stable version of 2.1, but not quite there yet. It was decided not to publish any more releases until we have a version that we are fairly confident of.

If you want, you can try setting up a version of Cura to run "from source". This is a bit hard, but not impossible.