Ultimaker/Cura #640: [2.099] Comprehensive list of machine configuration variables

CCS86, 2016-02-23:
Hi guys,

Do you have a list of all the valid machine configuration variables to set? (bed size, heated bed, axis layout,etc)

Thanks!

nallath, 2016-02-24:
Have a look at fdmprinter.json. That should contain all settings that can be changed.

Ghostkeeper, 2016-02-24:
In particular, the settings that are to be set by the machines are in the `machine_settings` category. The settings in that category will not be displayed in the interface.

CCS86, 2016-02-25:
Awesome, thanks @nallath and @Ghostkeeper

CCS86, 2016-02-26:
Are there any plans to have a "custom" option in the create a machine dialog, which gives plain text descriptions of these options via check box / drop down / input, to help the user build a machine cfg file?

fieldOfView, 2016-02-26:
It is not on the roadmap AFAIK, but is fairly "doable" when implemented like the current wizard for the UMO and UM2. Could be a nice community-project...

CCS86, 2016-04-07:
One more question guys:

How should the machine instance CFG file differ in format, compared to the JSON file?

Do I just drop the quotation marks and colons, and use equal signs instead?

`{
    "id": "prusa_i3",
    "version": 1, 
    "name": "Prusa i3",
    "manufacturer": "Other",
    "author": "Other",
    "icon": "icon_ultimaker2.png",
    "platform": "prusai3_platform.stl",
    "file_formats": "text/x-gcode",
    "inherits": "fdmprinter.json",

    "machine_settings": {
        "machine_heated_bed": { "default": true },
        "machine_width": { "default": 200 },
        "machine_height": { "default": 200 },
        "machine_depth": { "default": 200 },
        "machine_center_is_zero": { "default": false },
        "machine_nozzle_size": { "default": 0.4 },
        "material_diameter": { "default": 1.75 },
        "machine_nozzle_heat_up_speed": { "default": 2.0 },
        "machine_nozzle_cool_down_speed": { "default": 2.0 },
        "machine_head_shape_min_x": { "default": 75 },
        "machine_head_shape_min_y": { "default": 18 },
        "machine_head_shape_max_x": { "default": 18 },
        "machine_head_shape_max_y": { "default": 35 },
        "machine_nozzle_gantry_distance": { "default": 55 },
        "machine_gcode_flavor": { "default": "RepRap (Marlin/Sprinter)" },

        "machine_start_gcode": {
            "default": "G21 ;metric values\nG90 ;absolute positioning\nM82 ;set extruder to absolute mode\nM107 ;start with the fan off\nG28 X0 Y0 ;move X/Y to min endstops\nG28 Z0 ;move Z to min endstops\nG1 Z15.0 F9000 ;move the platform down 15mm\nG92 E0 ;zero the extruded length\nG1 F200 E3 ;extrude 3mm of feed stock\nG92 E0 ;zero the extruded length again\nG1 F9000\n;Put printing message on LCD screen\nM117 Printing..."
        },
        "machine_end_gcode": {
            "default": "M104 S0 ;extruder heater off\nM140 S0 ;heated bed heater off (if you have it)\nG91 ;relative positioning\nG1 E-1 F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure\nG1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more\nG28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way\nM84 ;steppers off\nG90 ;absolute positioning"
        }
    }
}
`

`[general]
name = Ultimaker Original
type = ultimaker_original
active_profile = Lars
version = 1
material = PLA

[machine_settings]
machine_heated_bed = True
machine_extruder_drive_upgrade = True

`

CCS86, 2016-04-07:
I am struggling to get this to work. With this machine CFG in place, Cura won't start. Here is the CFG and the Cura log:

```
[general]
name = AM1:4-7
type = fdmprinter
active_profile = AM1
version = 1
material = ABS

[machine_settings]
machine_heated_bed = true
machine_width = 1100
machine_height = 850
machine_depth = 900
machine_center_is_zero = false
machine_nozzle_size = 0.8
material_diameter = 1.75
machine_gcode_flavor = RepRap (Marlin/Sprinter)
machine_start_gcode = "G1 Z100 F500\nM204 S190 ;set accel value (150-250)\nM205 X0.005 Z.005"
machine_end_gcode = "M104 S0 ; turn off extruder\nM140 S0 ; turn off bed\nG28 Y0; Home Y\nM43\nM106 S0; Turn off fan"

```

```
2016-04-07 12:19:47,342 - INFO - Loaded plugin FileLogger
2016-04-07 12:19:47,348 - INFO - Loaded plugin GCodeProfileReader
2016-04-07 12:19:47,355 - INFO - Loaded plugin GCodeWriter
2016-04-07 12:19:47,364 - INFO - Loaded plugin ImageReader
2016-04-07 12:19:47,373 - INFO - Loaded plugin LayerView
2016-04-07 12:19:47,381 - INFO - Loaded plugin LegacyProfileReader
2016-04-07 12:19:47,390 - INFO - Loaded plugin LocalFileOutputDevice
2016-04-07 12:19:47,401 - INFO - Loaded plugin PerObjectSettingsTool
2016-04-07 12:19:47,414 - INFO - Loaded plugin PostProcessingPlugin
2016-04-07 12:19:47,428 - INFO - Loaded plugin RemovableDriveOutputDevice
2016-04-07 12:19:47,443 - INFO - Loaded plugin SliceInfoPlugin
2016-04-07 12:19:47,454 - INFO - Loaded plugin SolidView
2016-04-07 12:19:47,466 - INFO - Loaded plugin CameraTool
2016-04-07 12:19:47,480 - INFO - Loaded plugin MirrorTool
2016-04-07 12:19:47,496 - INFO - Loaded plugin RotateTool
2016-04-07 12:19:47,512 - INFO - Loaded plugin ScaleTool
2016-04-07 12:19:47,525 - INFO - Loaded plugin SelectionTool
2016-04-07 12:19:47,539 - INFO - Loaded plugin TranslateTool
2016-04-07 12:19:47,554 - INFO - Checking for new version of cura
2016-04-07 12:19:47,554 - INFO - Loaded plugin UpdateChecker
2016-04-07 12:19:47,577 - INFO - Loaded plugin USBPrinting
2016-04-07 12:19:47,592 - INFO - Loaded plugin SimpleView
2016-04-07 12:19:47,607 - INFO - Loaded plugin WireframeView
2016-04-07 12:19:47,623 - INFO - Loaded plugin XRayView
2016-04-07 12:19:47,623 - INFO - Command line arguments: {'debug-mode': False, 'disable-textures': False, 'file': [], 'external-backend': False}
2016-04-07 12:19:47,624 - DEBUG - Connected Boxes: {}
2016-04-07 12:19:47,893 - WARNING - No active machine found when trying to save setting visibility
2016-04-07 12:19:47,897 - DEBUG - Attempting to kill the engine process
2016-04-07 12:19:47,902 - ERROR - An error occurred in inherit function for wall_line_count: unsupported operand type(s) for -: 'float' and 'str'
2016-04-07 12:19:47,902 - ERROR - An error occurred in inherit function for wall_line_count: unsupported operand type(s) for -: 'float' and 'str'
2016-04-07 12:19:47,907 - ERROR - An error occurred in inherit function for wall_line_count: unsupported operand type(s) for -: 'float' and 'str'
2016-04-07 12:19:47,907 - ERROR - An error occurred in inherit function for wall_line_count: unsupported operand type(s) for -: 'float' and 'str'
2016-04-07 12:19:47,909 - ERROR - An error occurred in inherit function for brim_line_count: unsupported operand type(s) for /: 'float' and 'str'
2016-04-07 12:19:47,909 - ERROR - An error occurred in inherit function for brim_line_count: unsupported operand type(s) for /: 'float' and 'str'
2016-04-07 12:19:47,911 - ERROR - An error occurred in inherit function for wall_line_count: unsupported operand type(s) for -: 'float' and 'str'
2016-04-07 12:19:47,911 - ERROR - An error occurred in inherit function for wall_line_count: unsupported operand type(s) for -: 'float' and 'str'
2016-04-07 12:19:47,913 - ERROR - An error occurred in inherit function for brim_line_count: unsupported operand type(s) for /: 'float' and 'str'
2016-04-07 12:19:47,913 - ERROR - An error occurred in inherit function for brim_line_count: unsupported operand type(s) for /: 'float' and 'str'
2016-04-07 12:19:47,923 - ERROR - An error occurred in inherit function for brim_line_count: unsupported operand type(s) for /: 'float' and 'str'
2016-04-07 12:19:47,923 - ERROR - An error occurred in inherit function for brim_line_count: unsupported operand type(s) for /: 'float' and 'str'
2016-04-07 12:19:47,941 - ERROR - An error occurred in inherit function for brim_line_count: unsupported operand type(s) for /: 'float' and 'str'
2016-04-07 12:19:47,941 - ERROR - An error occurred in inherit function for brim_line_count: unsupported operand type(s) for /: 'float' and 'str'
2016-04-07 12:19:47,952 - ERROR - An error occurred in inherit function for brim_line_count: unsupported operand type(s) for /: 'float' and 'str'
2016-04-07 12:19:47,952 - ERROR - An error occurred in inherit function for brim_line_count: unsupported operand type(s) for /: 'float' and 'str'
2016-04-07 12:19:47,972 - ERROR - An error occurred in inherit function for brim_line_count: unsupported operand type(s) for /: 'float' and 'str'
2016-04-07 12:19:47,972 - ERROR - An error occurred in inherit function for brim_line_count: unsupported operand type(s) for /: 'float' and 'str'
2016-04-07 12:19:48,061 - INFO - Started engine process: C:\Program Files\Cura 2.1_3-18\CuraEngine.exe
2016-04-07 12:19:48,068 - DEBUG - Setting active view to SolidView
```

CCS86, 2016-04-07:
If I change the "type" to "ultimaker_original", Cura will load, but displays an error that it is unable to slice.

CCS86, 2016-04-07:
Commenting out this line: "machine_nozzle_size = 0.8", drops all the "inherit function" errors, but the application still won't load.

```
2016-04-07 12:51:19,397 - INFO - Loaded plugin ConsoleLogger
2016-04-07 12:51:19,404 - INFO - Loaded plugin CuraEngineBackend
2016-04-07 12:51:19,408 - INFO - Loaded plugin CuraProfileReader
2016-04-07 12:51:19,412 - INFO - Loaded plugin CuraProfileWriter
2016-04-07 12:51:19,436 - INFO - Loaded plugin Doodle3D
2016-04-07 12:51:19,443 - INFO - Loaded plugin 3MFWriter
2016-04-07 12:51:19,448 - INFO - Loaded plugin OBJReader
2016-04-07 12:51:19,454 - INFO - Loaded plugin OBJWriter
2016-04-07 12:51:19,460 - INFO - Loaded plugin STLReader
2016-04-07 12:51:19,466 - INFO - Loaded plugin STLWriter
2016-04-07 12:51:19,473 - INFO - Loaded plugin FileLogger
2016-04-07 12:51:19,480 - INFO - Loaded plugin GCodeProfileReader
2016-04-07 12:51:19,488 - INFO - Loaded plugin GCodeWriter
2016-04-07 12:51:19,497 - INFO - Loaded plugin ImageReader
2016-04-07 12:51:19,506 - INFO - Loaded plugin LayerView
2016-04-07 12:51:19,515 - INFO - Loaded plugin LegacyProfileReader
2016-04-07 12:51:19,524 - INFO - Loaded plugin LocalFileOutputDevice
2016-04-07 12:51:19,534 - INFO - Loaded plugin PerObjectSettingsTool
2016-04-07 12:51:19,547 - INFO - Loaded plugin PostProcessingPlugin
2016-04-07 12:51:19,560 - INFO - Loaded plugin RemovableDriveOutputDevice
2016-04-07 12:51:19,575 - INFO - Loaded plugin SliceInfoPlugin
2016-04-07 12:51:19,586 - INFO - Loaded plugin SolidView
2016-04-07 12:51:19,598 - INFO - Loaded plugin CameraTool
2016-04-07 12:51:19,612 - INFO - Loaded plugin MirrorTool
2016-04-07 12:51:19,627 - INFO - Loaded plugin RotateTool
2016-04-07 12:51:19,642 - INFO - Loaded plugin ScaleTool
2016-04-07 12:51:19,656 - INFO - Loaded plugin SelectionTool
2016-04-07 12:51:19,670 - INFO - Loaded plugin TranslateTool
2016-04-07 12:51:19,684 - INFO - Checking for new version of cura
2016-04-07 12:51:19,684 - INFO - Loaded plugin UpdateChecker
2016-04-07 12:51:19,707 - INFO - Loaded plugin USBPrinting
2016-04-07 12:51:19,723 - INFO - Loaded plugin SimpleView
2016-04-07 12:51:19,738 - INFO - Loaded plugin WireframeView
2016-04-07 12:51:19,754 - INFO - Loaded plugin XRayView
2016-04-07 12:51:19,754 - INFO - Command line arguments: {'file': [], 'disable-textures': False, 'external-backend': False, 'debug-mode': False}
2016-04-07 12:51:19,758 - DEBUG - Connected Boxes: {}
2016-04-07 12:51:20,018 - WARNING - No active machine found when trying to save setting visibility
2016-04-07 12:51:20,020 - DEBUG - Attempting to kill the engine process
2016-04-07 12:51:20,168 - INFO - Started engine process: C:\Program Files\Cura 2.1_3-18\CuraEngine.exe
2016-04-07 12:51:20,174 - DEBUG - Setting active view to SolidView
Traceback (most recent call last):
  File "cura_app.py", line 30, in <module>
  File "C:\Users\ultimaker\Desktop\cura-build\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 185, in run
  File "C:\Users\ultimaker\Desktop\cura-build\build\inst\lib\python3.4\site-packages\cura\BuildVolume.py", line 43, in __init__
  File "C:\Users\ultimaker\Desktop\cura-build\build\inst\lib\python3.4\site-packages\cura\BuildVolume.py", line 173, in _onActiveInstanceChanged
  File "C:\Users\ultimaker\Desktop\cura-build\build\inst\lib\python3.4\site-packages\cura\BuildVolume.py", line 81, in rebuild
TypeError: bad operand type for unary -: 'str'

```

CCS86, 2016-04-07:
Okay, I have narrowed it down to the following:

Calling this setting:

machine_nozzle_size = 0.8

Is causing this error:
[image]

And these settings are preventing the front end from loading:

machine_depth = 900
machine_height = 850
machine_width = 1100

Ghostkeeper, 2016-04-12:
Looks like we're doing calculations with the pre-evaluated string directly from the JSON for machine_nozzle_size, rather than evaluating these values first. Thanks for the report, we'll check it out. I'm not sure if this can be fixed for 2.1 though. It might have to wait for 2.2.

awhiemstra, 2016-04-12:
The problem here is that machine settings are not really settings at the moment. This will change for 2.2 so will be fixed then.

CCS86, 2016-04-12:
Is there a workaround for the time being? 

Can I just duplicate the fdmprinter.json, rename, and change these specific settings?

CCS86, 2016-04-12:
I tried duplicating the fdmprinter.json, renaming the description/type, changing all of the 0.4mm defaults to 0.8, and changing the build volume dimensions. I then created a new machine using that template, but it still doesn't work.

```
2016-04-12 12:11:56,751 - DEBUG - Connected Boxes: {}
2016-04-12 12:11:56,767 - INFO - Loaded plugin XRayView
2016-04-12 12:11:56,767 - INFO - Command line arguments: {'disable-textures': False, 'external-backend': False, 'file': [], 'debug-mode': False}
2016-04-12 12:11:56,942 - WARNING - No active machine found when trying to save setting visibility
2016-04-12 12:11:56,943 - DEBUG - Attempting to kill the engine process
2016-04-12 12:11:56,948 - WARNING - Tried to set value of non-user setting brim_line_count
2016-04-12 12:11:56,949 - ERROR - An error occurred in inherit function for wall_line_count: unsupported operand type(s) for -: 'float' and 'str'
2016-04-12 12:11:56,949 - ERROR - An error occurred in inherit function for wall_line_count: unsupported operand type(s) for -: 'float' and 'str'
2016-04-12 12:11:56,958 - WARNING - Tried to set value of non-user setting brim_line_count
2016-04-12 12:11:56,977 - WARNING - Tried to set value of non-user setting brim_line_count
2016-04-12 12:11:56,987 - WARNING - Tried to set value of non-user setting brim_line_count
2016-04-12 12:11:57,006 - WARNING - Tried to set value of non-user setting brim_line_count
2016-04-12 12:11:57,043 - WARNING - Could not determine application version from string 2.1.0-20160318, not checking for updates
2016-04-12 12:11:57,058 - WARNING - Tried to set value of non-user setting brim_line_count
2016-04-12 12:11:57,060 - ERROR - An error occurred in inherit function for wall_line_count: unsupported operand type(s) for -: 'float' and 'str'
2016-04-12 12:11:57,061 - ERROR - An error occurred in inherit function for infill_line_distance: unsupported operand type(s) for /: 'str' and 'int'
2016-04-12 12:11:57,063 - ERROR - An error occurred in inherit function for support_roof_line_distance: unsupported operand type(s) for /: 'str' and 'int'
2016-04-12 12:11:57,064 - ERROR - An error occurred in inherit function for support_line_distance: unsupported operand type(s) for /: 'str' and 'int'
2016-04-12 12:11:57,071 - INFO - Started engine process: C:\Program Files\Cura 2.1_3-18\CuraEngine.exe
2016-04-12 12:11:57,076 - DEBUG - Setting active view to SolidView
Traceback (most recent call last):
  File "cura_app.py", line 30, in <module>
  File "C:\Users\ultimaker\Desktop\cura-build\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 185, in run
  File "C:\Users\ultimaker\Desktop\cura-build\build\inst\lib\python3.4\site-packages\cura\BuildVolume.py", line 43, in __init__
  File "C:\Users\ultimaker\Desktop\cura-build\build\inst\lib\python3.4\site-packages\cura\BuildVolume.py", line 173, in _onActiveInstanceChanged
  File "C:\Users\ultimaker\Desktop\cura-build\build\inst\lib\python3.4\site-packages\cura\BuildVolume.py", line 151, in rebuild
  File "C:\Users\ultimaker\Desktop\cura-build\build\inst\lib\python3.4\site-packages\cura\BuildVolume.py", line 262, in _getSkirtSize
TypeError: can't multiply sequence by non-int of type 'NoneType'

```

Ghostkeeper, 2016-04-13:
I could replicate your problem, but I have no good fix.

I made this patch for you: http://dulek.net/work/BuildVolume.py.patch
It basically puts "float()" or "int()" around all the settings in BuildVolume that need to be numbers. That fixes the issue itself and lets Cura start with your large build volume (the buildplate mesh protrudes the buildplate now but it's workable). However, this does seem to break some inheritance functions for some reasons so I won't commit it. We'll have to make a proper fix for version 2.2.

awhiemstra, 2016-04-13:
The best workaround is to create your own machine definition instead of trying to do things through the machine instance. As long as you inherit from fdmprinter you should get all settings. You can specify any overrides you want in the "overrides" bit of the machine definition. Take a look at the other machine definitions for how to use that.

CCS86, 2016-04-13:
@awhiemstra , I thought that's what I was doing by duplicating the fdmprinter.json and renaming/customizing it. How can you get around creating a machine instance? That seems to happen automatically.

I tried deleting all my machine instances and copying my modified JSON from the: C:\Program Files\Cura 2.1_3-18\resources\machines directory, over to: User\AppData\Local\cura\machines

When I launch Cura, it still prompts to create a machine, and after dismissing that dialog, my machine was not available in the machine selection drop down.

CCS86, 2016-04-13:
@Ghostkeeper, thank you for working on that for me. Can you point me in the right direction on how to apply the patch?

Ghostkeeper, 2016-04-14:
I do suggest taking @awhiemstra's approach, but try copying the definition for the Ultimaker Original instead of FDMPrinter. When you create a machine definition (in \cura\machines) you define a machine but you'd still have to "add" the machine. So don't dismiss the dialogue to create a machine. Rather, select your new machine and add it. Cura will create the machine instance then.

To apply my patch, you'd have to run Cura from source, because py2exe packs all the Python file in an executable, preventing you from modifying it. Download the patch and put it in your git repository. Navigate to your git directory in a console and type `git apply BuildVolume.py.patch`. That should apply the patch to the correct file.

Ghostkeeper, 2016-05-26:
Fixed by the settings revamp that was done.

closed 2016-05-26 by Ghostkeeper