Ultimaker/Cura #644: ImportError: No module named 'UM.View.GL'

danilius, 2016-02-26:
I'm on Ubuntu 15.04, and got this:

Traceback (most recent call last):
  File "./cura_app.py", line 20, in <module>
    import cura.CuraApplication
  File "/media/dani/flyyngga/apps/cura_build/Cura/cura/CuraApplication.py", line 37, in <module>
    from . import PlatformPhysics
  File "/media/dani/flyyngga/apps/cura_build/Cura/cura/PlatformPhysics.py", line 19, in <module>
    from . import ConvexHullJob
  File "/media/dani/flyyngga/apps/cura_build/Cura/cura/ConvexHullJob.py", line 10, in <module>
    from . import ConvexHullNode
  File "/media/dani/flyyngga/apps/cura_build/Cura/cura/ConvexHullNode.py", line 10, in <module>
    from UM.View.GL.OpenGL import OpenGL
ImportError: No module named 'UM.View.GL'

...and of course Cura won't start. Googled and tried what I found with no result.

fieldOfView, 2016-02-26:
Did you also clone the Uranium repository and did you add it to your PYTHONPATH env variable?

danilius, 2016-02-26:
Yup.

thopiekar, 2016-02-26:
Don't see this problem here in my personal nightly builds.

nallath, 2016-02-29:
The error you have is something with regards to your pythonpath.

danilius, 2016-02-29:
Typing echo $PYTHONPATH in the terminal gives:

:/media/dani/flyyngga/apps/cura_build/Uranium

Is this correct? Have I done something silly?

thopiekar, 2016-02-29:
If /media/dani/flyyngga/apps/cura_build/Uranium is the Uranium package then
it is wrong. Let's say you want to import "example, then PYTHONPATH should
be the directory where the "example" is in.

I guess PYTHONPATH=/media/dani/flyyngga/apps/cura_build/ should be correct
:)

danilius, 2016-02-29:
Thanks, will give that a whirl

danilius, 2016-02-29:
Hmm. Seems I am thicker than I thought. 

Cura and Uranium sit inside the same folder, called cura_build. Is that correct, or should Uranium be somewhere inside the Cura folder?

danilius, 2016-02-29:
What I find confusing is that @fieldOfView wrote:
> Did you also clone the Uranium repository and did you add it to your PYTHONPATH env variable?

That suggests (to me anyway) that the PYTHONPATH environment variable should point towards Uranium. Anyway, I pointed it at the parent folder, cura_build, and got the same result.

nallath, 2016-02-29:
I run cura in the following way on Linux;

PYTHONPATH=.:/usr/local/lib/python3/dist-packages/:../Uranium/ python3 cura_app.py

As you can see I explicitly tell it where to find Uranium

thopiekar, 2016-02-29:
There are two different ways to build Cura. First there are build scripts
in a separate repository and you can also build it on you own, by cloning
every repository (Arcus, Uranium, CuraEngine and Cura). I think
@fieldOfView believed you just cloned everything manually. So in case
you've done this you'll get a directory with the cloned repository. This
would result into /build-dir/Uranium/Uranium where the last Uranium is your
package and /build-dir/Uranium/ is your PYTHONPATH (you will need to do the
same for Arcus, etc.).

I personally neither use the build scripts nor the manual build method, I
just build Cura as nightly builds in a personal repository for Ubuntu. Hope
the info will be useful for your needs.

danilius, 2016-02-29:
Ah, so the plot thickens. 

All I did was clone Cura and Uranium into a folder called cura_build. Do I have to clone Arcus and CuraEngine as well?

thopiekar, 2016-02-29:
I never tried without but as far as I know you'll first need Arcus, then
build with it CuraEngine (enable Arcus usage there!). Additionally build
Uranium over them and finally build Cura. All these 4 sources are needed
(at least they are built in my case).
Possibly something got not built because Arcus was missing ;)

danilius, 2016-02-29:
OK, I'm going to try the script, but as I have Cura 15.04 installed, will this overwrite the CuraEngine that is already installed, or will it all build in the local directory? I don't want to hose my existing installation of CuraEngine.

Ghostkeeper, 2016-03-01:
cura_build will build it all in a local directory. Except that you can specify the install directory in the CMake script too, and if you run `make install` then it will install CuraEngine (among others) to that directory and might overwrite it. I suggest you try just `make` first and try to run it locally.

danilius, 2016-03-01:
Gonna do just that, thanks!

thopiekar, 2016-03-09:
I guess it worked ^^

danilius, 2016-03-09:
Nope, it bombed out with some other problem and I don't have the time to faff around with it just now, sadly.

closed 2016-12-26 by thopiekar

venkateshdhara, 2017-10-22:
when i run setup.py i was getting following error in python IDLE
    import UM
ModuleNotFoundError: No module named 'UM'

Ghostkeeper, 2017-10-30:
Is the UM folder of Uranium in your PYTHONPATH environment variable?

venkateshdhara, 2017-11-04:
that problem was solved 
when i try to build setup.py following errors were getting display in terminal window
I:\Cura-master>python setup.py build
Removing previous distribution package
running build
running build_scripts
Coping Cura plugins.
Coping resources.
Coping Uranium QML.
Coping PyQt5 plugins from: C:\Users\venky\AppData\Local\Programs\Python\Python36
-32\lib\site-packages\PyQt5
Traceback (most recent call last):
  File "setup.py", line 67, in <module>
    shutil.copytree(os.path.join(qt_origin_path, "plugins"), "dist/PyQt5/plugins
")
  File "C:\Users\venky\AppData\Local\Programs\Python\Python36-32\lib\shutil.py",
 line 309, in copytree
    names = os.listdir(src)
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\
Users\\venky\\AppData\\Local\\Programs\\Python\\Python36-32\\lib\\site-packages\
\PyQt5\\plugins'

Please guide i think these errors are regarding some path settings ? Thank you

Ghostkeeper, 2017-11-10:
To install PyQt, I'd suggest this command:

    python -m pip install PyQt5

venkateshdhara, 2017-12-22:
DO i need to install any plugins related to PyQt5 ?

LipuFei, 2017-12-22:
@venkateshdhara You don't need any plugins, and if you just want to run from source code, you don't need to run `python setup.py build` for Cura and Uranium as well.

hardix02, 2023-06-28:
> Did you also clone the Uranium repository and did you add it to your PYTHONPATH env variable?

i am getting Error : No module named 'pyArcus' what i do?