Ultimaker/Cura #645: [2.1] Empty status message when writing to write-protected SD-card.

thopiekar, 2016-02-26:
I wondered about this behavior and just noticed that I probably played with the write-protection switch on my SD-card.

dmesg:
```
[27423.002635] scsi host7: usb-storage 1-1.5:1.0
[27424.003727] scsi 7:0:0:0: Direct-Access     Generic- Multi-Card       1.00 PQ: 0 ANSI: 0 CCS
[27424.004214] sd 7:0:0:0: Attached scsi generic sg2 type 0
[27424.681734] sd 7:0:0:0: [sdb] 7788544 512-byte logical blocks: (3.98 GB/3.71 GiB)
[27424.682828] sd 7:0:0:0: [sdb] Write Protect is on
[27424.682834] sd 7:0:0:0: [sdb] Mode Sense: 03 00 80 00
[27424.683866] sd 7:0:0:0: [sdb] No Caching mode page found
[27424.683873] sd 7:0:0:0: [sdb] Assuming drive cache: write through                                          
[27424.688570]  sdb: sdb1
[27424.692201] sd 7:0:0:0: [sdb] Attached SCSI removable disk
```

mount:
```
/dev/sdb1 on /media/thopiekar/HEPHESTOS 2 type vfat (ro,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
```

Screenshot:
[image]

nallath, 2016-02-29:
Nice find. So if I read this correctly, cura tried mounting a write protected card (which failed, but wasn't caught correctly?)

thopiekar, 2016-02-29:
Nope, Cura is not able to mount a SD card here on Linux (as far as I know).
I already mounted the SD card with a desktop widget before. As the
read-only switch on the SD card was on, Linux (as any other OS) is not able
to mount it with writing permissions. When trying to write to the SD card
there is the empty message you see.
It is also possible that there is a bug with the translations (using German
translations here).

Ghostkeeper, 2016-02-29:
It is either not just the German translation or it is also in the English translation, since it also happens on my computer if I switch the SD card to locked state. I've seen it before.

MarcusWolschon, 2016-05-01:
this is quite important if you have SD cards with a loose write protection switch that may move into the read only state by the process of inserting it into the card slot.

thopiekar, 2016-05-01:
This is how I ran into this problem. The cardreader of my notebook often likes to push the SD card into read-only mode 😌

awhiemstra, 2016-05-02:
The message should state that it could not write to the SD card because of a permission error. For some reason it does not, we already have an internal issue for it, it should be a minor fix.

MarcusWolschon, 2016-06-16:
Bug still present in 2.1.1

Ghostkeeper, 2016-06-16:
Hmm, this was supposedly fixed by https://github.com/Ultimaker/Cura/commit/6237767ba5841ce7ef8eaea1b539bac4ffa7d104. It was tested by me and Andreea and confirmed it works on each supported operating system. Perhaps 2.1.1 was before the fix?

nickthetait, 2016-06-17:
There is a tag for 2.1.0 but not for 2.1.1 or 2.1.2, so I can't tell if that commit is included or not.

thopiekar, 2016-06-17:
@nickthetait Very simple. Click on the commit, click on "View" button and, if you are still lazy, click on "History" and you land here: https://github.com/Ultimaker/Cura/commits/6237767ba5841ce7ef8eaea1b539bac4ffa7d104/plugins/RemovableDriveOutputDevice/RemovableDriveOutputDevice.py

So, yes, it is the last change on this file.

thopiekar, 2016-06-17:
Oh and for 2.1:
https://github.com/Ultimaker/Cura/commits/2.1/plugins/RemovableDriveOutputDevice/RemovableDriveOutputDevice.py
and master:
https://github.com/Ultimaker/Cura/commits/master/plugins/RemovableDriveOutputDevice/RemovableDriveOutputDevice.py

Therefore: Only included in master :wink:

nickthetait, 2016-06-17:
@thopiekar that is a handy github trick to know. Thanks!

thopiekar, 2016-06-18:
@nickthetait No problem :+1: 

@Ghostkeeper: Do you think it will be save to cherry-pick this into [2.1]?

Ghostkeeper, 2016-06-20:
It'd probably be safe, but it's not a workflow blocking bug, just a display issue.

BagelOrb, 2017-07-21:
Probably fixed.
We take care of write protection on SD cards, right?

closed 2017-07-21 by BagelOrb

Ghostkeeper, 2017-07-27:
Yes, it was fixed in 2.3.