Ultimaker/Cura #647: Start/end Gcode scripts editor missing.

guigashm, 2016-02-27:
Hello,

The old CURA had a tab where you can add gcode scripts (ex: Start / End). On the "new GUI" this is missing. 

It's a bug? or there is a button somewhere? (i'm using MAC OS X build).

To be able to do this, I went to the Machine dir and edited the .json.

But this is not very practical and user friendly.

thank you

nallath, 2016-02-29:
There is no "easy" gui part where you can change start / end g-codes yet.

mastory, 2016-03-10:
Please replace this feature similar to the legacy Cura.  Thanks

nallath, 2016-03-11:
It's not going to be in the next release, that's for sure. 
Can't say if it will be in the one after that, but if someone is willing to build it, I'm more than willing to give them some pointers & info on how to do it best.

brainscan, 2016-07-12:
even if you do edit the start/end code cura doesn't follow it and adds in some of its own moves which is ridiculous and potentially dangerous. I have to re edit the code it gives me before I can use it.

Ghostkeeper, 2016-07-13:
@brainscan Are you sure you haven't got a post-processing script active?
The start/end g-code is added all the way at the end of the process. Only the post-processing scripts are able to modify it (to put moves in between). The start/end g-code otherwise just gets inserted as one block of characters.

brainscan, 2016-07-13:
Does cura have a built in post processor? How would I switch that on and off? I've changed the gcode in .json files but that isn't the only place code seems to get added.

fieldOfView, 2016-07-13:
Here is a first version of a plugin that allows you to edit start- and end-gcode, as well as print volume dimensions:
https://github.com/fieldOfView/MachineSettingsActionPlugin
It requires Cura 2.2, which has no official release yet. You would have to run Cura from source to test it now.

Note that the plugin does not currently like empty lines or lines that start with a semi-colon. The plugin has rough edges.

Ghostkeeper, 2016-07-13:
The post processor is by default included in our builds. It's under Extensions -> Post Processing -> Modify G-code. If you don't have it in your build, you can download it from here: http://github.com/nallath/PostProcessingPlugin

But if you didn't know about that one, I expect you didn't set a post processing script yet :stuck_out_tongue: In that case, we might have a new bug on our hands though I can't imagine how that could happen in 2.1.

brainscan, 2016-07-13:
> But if you didn't know about that one, I expect you didn't set a post processing script yet 😛 In that case, we might have a new bug on our hands though I can't imagine how that could happen in 2.1.

I haven't used any extensions, I didn't realise the code would need post processing as well.

BagelOrb, 2017-07-21:
implemented

closed 2017-07-21 by BagelOrb