Ultimaker/Cura #648: First run wizard and machine settings, bed size,nozzle and filament size missing.

guigashm, 2016-02-27:
Hello,

Looking at the "old Cura", something is missing.

First run wizard (machine selection) bed size, nozzle and filament size missing.

Also on machine profile settings windows, near the "add" button that should be an edit button, to change the existing machine settings (like bed size, nozzle, filament size).

Or at list, the Machine drawer on advanced, should have this option

thank you,

nallath, 2016-02-29:
The bed-size is now set by the type of machine you select. Same goes for nozzle & filament size. 

Changing the machine can be done by changing the machine files. If your machine isn't in the list, make the file and then contribute it back so others don't have to do this anymore.

guigashm, 2016-03-02:
Hello nallath,

I did contribute and post Prusa I3 XL.json and .STL. And the Pull request was accepted.

But that's not the point. The point is simplification and better maintenence.

There are printers, that the model is the same but, they can have filament 1,75 or 3 mm  (or  different  bed sizes)..
If you start creating different machine .jsons for each printer you will have a never ending list of printers!

Every slicing software has this.. and also was in the "old Cura".

At least put this as an Enhancement!

Thank you,

fieldOfView, 2016-03-02:
I don't agree that filament size is a machine property either. Filament size is a material property! Even an Ultimaker printer, which is a 3 mm printer, can print with 2.85mm filament or filament that is actually 3mm. It would be "wrong" to have to compensate for this with the flow setting.

nallath, 2016-03-03:
You make a fair point with regards to the 2.85 vs the 1.75 mm fillament.

I don't see a lot of issues with having a ton of .json files. Most 3D printers have a 'fixed' size of things (eg; Bed & buildplate size are always the same). 

Having the same model but only one thing changed is very easy to build. Simply use inheritance. A good example of this can be seen in the UM2 Extended, which is a UM2 except that we use overrides to change the size of the machine.

That the old cura did something in a certain way is no reason for me to implement it.

Netoperz, 2016-03-24:
I do not agree, i have custom printer for my own use I want to use cura, normally there was, type of machine - other, and settings "custom"  where i could configure the settings.

For example that was normal operation for Reenforce RF2000 printer that I do own.
So now i cannot use cura.

Is that change made to cut off straight support for other makers printers (other than ultimaker) ?

For example i do have manual for my printer https://www.conrad .de/de/renkforce-rf2000-3d-drucker-dual-extruder-1395717.html  file  : 

http://www.produktinfo.conrad.com/datenblaetter/1300000-1399999/001395717-an-01-de-RENKFORCE_3D_DRUCKER_RF2000.pdf

And there is a process off adding setting to cura. 
Now i cannot do that. So Must use Repeater, they do not change things like that.
Or use old version of cura. 

That is not nice IMHO.
The custom settings creator was very useful.
It's petty that after upgrade and reinstall i need to switch to repeater and stop using cura.

fieldOfView, 2016-03-24:
@Netoperz calm down, there is no conspiracy here.

Cura - both the old version and the totally rewritten from scratch version we're talking about here - are open source, but funded by Ultimaker. It is not all that strange that Ultimaker is focussing on getting the software they are writing right for their printers first. But even now, there is support for 3rd party printers. If your printer is not one of the printers we have a standard profile for, you can make a profile of your own:
https://github.com/Ultimaker/Cura#making-profiles-for-other-printers
Chances are that along the way Ultimaker, or someone from the community, makes a plugin to do that from within Cura. But for now the Ultimaker staff has more important issues to work on.

Note that we're proud you use our software, but we have no obligation towards Conrad or what their manual says. If the old Cura works for you, by all means keep using that; there is not much that you can do with the new Cura that you cannot do with the old Cura right now. In the future, we hope to have a Cura that does (almost) everything the old Cura does and more.

closed 2016-03-24 by fieldOfView

Ghostkeeper, 2016-03-24:
The idea was that anyone who's able to build a custom printer is also able to edit a .json file in Cura's machines folder. A wizard to add a custom machine with different dimensions is planned, but with no significant priority because of this.

fieldOfView, 2016-03-24:
To add to that, Conrad can make a profile for the machine and either make it available for download or send it in to have it included in Cura by default.

Netoperz, 2016-03-24:
Sorry, i'm a bit pissed off because lately many things have changed in my software development platform, faulty laptop, and other mess and at the end i wanted to printout the prototype after clean install, and one more trouble with setting up cura (writing the profile) 

For now i will use repetier because it has functions for manual configuration without complicating the process, the functionality that new cura does not have anymore.

I'll try to find a moment to make the configuration file and i'll make pul push request to add the profile for that printer, waiting for conrad electronic to do something about that is waiting for the rain on the desert. So i will contribute if i can (i'm also active open source developer, so i care about those who help others, and I'm trying to help always if i can) Sorry for "bad tone" but sometimes lack of time is making a frustration :D... Happy Easter :D to all.

fieldOfView, 2016-03-24:
We all have our bad days ;-) Out of curiosity: Any reason you can't use Cura 15.04.04 ("the old Cura")?

Netoperz, 2016-03-24:
I think that i can use it, just downloaded "current" and there was the problem.

The thing is that, if i will need reinstall or tell to my clients to use cura, word "old" will not be good marketing practice, people want "new" , "actual" etc... and as i'm designing new printer to be run as retail product, would be nice to have cura regardless of version to be compatible.

But i think that for testing i can do things manualy, and if the device I'm designing will hit the market (next month) i will make a production profile and ask You to add to the recent build.

And for the RF2000 compatibility it should not be a problem. I do own RF2000 as a tool in my office.

Main reason for adding the profiles is to make it faster to reinstall software in the future. There is a point in the way that your team choose. And i do agree that adding profiles is a nice feature. But in my opinion it will decrease  the number of users that are experimenting/testing/building stuff.
It a bit complicates the initial development of the hardware, so IMHO if someone will start a new hardware project , that person/team will choose software that lets to change settings rapidly, and then that software will stay as mainstream software while cura may be alternative. 

But that is just pure divagation about "what if" :)

fieldOfView, 2016-03-24:
> divagation

I can honestly say that I had never seen that word, or the Latin word it is derived of, in my life. And my life is now richer because of it. Happy easter ;-)

Netoperz, 2016-03-24:
http://www.thefreedictionary.com/divagation

:) just a normal word I'm using quite recently, but what can I say... I'm a strange person maybe :) second thing is that English is not my native language (but i'm using it more than my own) third and last thing is that there is a bunch of words that are very sexy I think, and not much used by masses , that it is wort to know them :) Yap happy Easter :)

Ghostkeeper, 2016-03-25:
I'm going to mark this as a duplicate of #223.