Ultimaker/Cura #649: Bottom layer speed should be variable for lines and circles

thhart, 2016-02-29:
The bottom layer speed should be different for lines and other complex geometries like circles. The problem is that the filament (eg. PLA) does not adhere when printing small circles or curves but it does well with straight lines. It would be nice to avoid to slow down the printing speed for only some curves for the whole layer.

Ghostkeeper, 2016-02-29:
You're suggesting to use some sort of acceleration term then. Or maybe have the speed be related to the length of the extrusion move.

thhart, 2016-03-15:
I don't know if the suggested fix is suitable, however I will try to explain it different: When printing M4 holes for screws or small geometric structures and circles I have always adhesion problems in the very first layer on the bed. The printer is printing the inner diameter as first and the filament can hardly adhere. I have to print extremely slowly (12) to avoid this. Attached is a photo illustrating the problem, you can see a lot of circles and rectangles. Some are printed ok, some failed at least for the inner shell. It looks like the printer is still to fast for these small diameters and distances and the filament has no time to adhere.
[image]

nallath, 2016-03-16:
Seems like there is also backlash going on on your machine.

BagelOrb, 2017-07-21:
Perhaps an initial layer acceleration would help a lot!

aalku, 2018-07-07:
In my case this is not a problem only in first layer. I would need small circles to be printed slower on every layer.

smartavionics, 2018-07-07:
There is currently a PR (https://github.com/Ultimaker/Cura/pull/3837) that provides a similar capability. It has yet to be accepted.

closed 2018-11-26 by Appesteijn