Ultimaker/Cura #650: Cura crash-after loading Stl_does not complete theslice

twizter, 2016-02-29:
Cura crash-after loading Stl_does not complete the slice
End programme pop up details below

Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	CuraEngine.exe
  Application Version:	0.0.0.0
  Application Timestamp:	55b6398c
  Fault Module Name:	msvcrt.dll
  Fault Module Version:	7.0.7600.16385
  Fault Module Timestamp:	4a5bda6f
  Exception Code:	c0000005
  Exception Offset:	000143df
  OS Version:	6.1.7600.2.0.0.256.1
  Locale ID:	1033
  Additional Information 1:	0a9e
  Additional Information 2:	0a9e372d3b4ad19135b953a78882e789
  Additional Information 3:	0a9e
  Additional Information 4:	0a9e372d3b4ad19135b953a78882e789

Ghostkeeper, 2016-03-01:
Are you at liberty to send the STL file and/or the settings you sliced it with? And what Cura version did you use to slice it?

That log provides us with little useful information, I'm afraid :disappointed:

BagelOrb, 2017-07-21:
no info provided

closed 2017-07-21 by BagelOrb