Ultimaker/Cura #655: Feature Request: intermediate solid layers to increase radial strength?

williamsgray, 2016-03-08:
Might it be possible to add an infill? option for the number of intermediate solid layers?  (Solid "layer" thickness could default to top surface thickness, or optionally be specified here also.)   
For example, an object that takes 1000 layers to print, with 3 intermediate "solid layers" requested, would have a 100% fill section around layers 250, 500, and 750.    
I'm hoping something like this might provide a way to increase radial strength, without requiring more dense fill throughout.   Perhaps there are better ways to do this, or to specify this?   Thoughts?
Thanks!

markwal, 2016-03-08:
Print Settings->Infill->Advanced: Solid infill every n layers.

Since you have that, probably the thing you want is to say how many solid layers every n layers, currently only one.

You might also want to look at 3D infill patterns, but...

Apparently, for many models you'll get more strength for the same amount of plastic by adding more perimeters and thinking of infill more like internal support for top layer bridging.

Ghostkeeper, 2016-03-09:
How much would a single solid layer really contribute to strength? In my prints (though depending on the infill density) it often takes 2 or 3 layers for the top skin to fully close.

daid, 2016-03-09:
Some people actually "cut" 0.1mm cylinders trough their model to add extra material to strenghen up parts. This might help as a workaround.

ianpaschal, 2018-05-24:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

closed 2018-05-24 by ianpaschal