Ultimaker/Cura #656: Does not slice under Wine.

iHateInventNames, 2016-03-09:


thopiekar, 2016-03-09:
Why not installing Cura correctly from source or from prebuilt packages, like *.deb for Ubuntu or *.dmg for OSX??

nallath, 2016-03-09:
What Thopiekar said. I don't see any reason as to why running under wine should be at all possible.

closed 2016-03-09 by nallath

iHateInventNames, 2016-03-09:
Because I use Gentoo Linux

thopiekar, 2016-03-09:
I still see no reason why you need wine. Quickly googled and found https://packages.gentoo.org/packages/dev-python/PyQt5 and I guess it will be easy for you to install all the other dependencies and build Cura from source, eg. with https://github.com/Ultimaker/cura-build you can find here.
The only (problematic) thing I see is to install protobuf v3 (only v2 here: https://packages.gentoo.org/packages/dev-libs/protobuf). You can find a working and tested version of protobuf v3 also here at https://github.com/Ultimaker/protobuf

Wine is a great suite of software, but in my point of view it makes only sense when trying to run closed-source software or non-Linux software. Really, try to build it from source. If you had Ubuntu I would prefer to use my PPA, which provides and very easy install and recent builds.

daid, 2016-03-09:
I see no reason why Wine should give problems. However, we are currently experiencing problems with 32bit windows that we are still investigating. That might explain things.

iHateInventNames, 2016-03-09:
Is it possible to download 64bit version for Windows?

Ghostkeeper, 2016-03-10:
Are you getting an error message when running under Wine?

You are using a very old version by the way.

iHateInventNames, 2016-03-11:
This one is newest beta I can get from official website.