Ultimaker/Cura #657: [2.1] USB printing - TypeError: requestWrite() takes from 2 to 3 positional arguments but 4 were given

thopiekar, 2016-03-09:
When clicking on the "print over USB"-button, I get this in the log:
```
2016-03-09 14:55:21,035 - DEBUG - Backend connected on port 49674
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 225, in event
    return super().event(event)
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 171, in event
    event._function_event.call()
  File "/usr/lib/python3/dist-packages/UM/Event.py", line 151, in call
    self._function(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/OutputDeviceManagerProxy.py", line 114, in _writeToDevice
    device.requestWrite(node, file_name, filter_by_machine)
TypeError: requestWrite() takes from 2 to 3 positional arguments but 4 were given
```

Appears here on builds based on the codebase from last weekend. Sorry, if it is already fixed.

Ghostkeeper, 2016-03-10:
Fixed I hope with this commit: https://github.com/Ultimaker/Cura/commit/3690eafb721bd722ecb462d6afc6bda748c04fe4

closed 2016-03-10 by Ghostkeeper

thopiekar, 2016-03-10:
Ok, will give it a try. Thanks :+1: