Ultimaker/Cura #661: Feature Request: Ability to play a sound when a print is complete

starr0stealer, 2016-03-11:
The end user can have the option to define a sound to be played once a print is completed. There can be built in sounds to select from, or allow the user to pick a sound file on the system. For printers with a heated bed the sound could be delayed until a target temperature is met (or multiple sounds can be played during each stage).

This would assist with longer prints that the user doesn't stay by the printer during the entire job, providing a notification that could be heard from a distance.

Could be handled via the plugin system (once its expanded to handle UI or other features outside of post-processing) or within the core software (preferred option as this would happen sooner than later).

nallath, 2016-03-11:
This can quite easily be done. The output device has a completedWriting signal; so you can hook up to that and play a sound. Shouldn't be to hard to write a plugin that does that.

I don't think that anyone from Ultimaker will spend time on this though. But as usual; If someone from the comunity / somehwere else / whatever wants to make this, I'd be happy to help you.

starr0stealer, 2016-03-11:
I must not be looking at the right plugin docs, initially I was going to do this myself but found that it seemed all the post-processing was to modify the resulting GCode. I also wasn't able to locate a plugin that did anything but that.

http://wiki.ultimaker.com/How_to_write_a_Cura_plugin

Could you point me in the direction of more information about the "completedWriting signal" you spoke of

starr0stealer, 2016-03-11:
I done some better research on the topic of plugins, I've found some information within the Uranium project, which I found that the OutputDevice module has a writeFinished, which I believe is what you are referring to.
So far I am still thinking this would be more ideal in the core software rather than a plugin. I want to offer more than just playing a sound, ie options on if the user even wants this and what sound to play, as well as target moments to play the sound such as bed temp. These settings would go in either preferences or the export config window, as well as possibly a checkbox on the printer control window (this feature would only make sense for USB printing).
I don't know if all of that is possible with a plugin, or if it would require taking a slightly longer route to achieve the same UX.

https://github.com/Ultimaker/Uranium/wiki/Creating-plugins

I am guessing the other information I found was before the Cura rewrite from the older "daid" version (Something I was completely unaware of before).

I will need to do quite a bit more research on my end. I actually use the Lulzbot Cura edition, and apparently it's based on the legacy version and a lot has changed since. I suppose in the end if I am able to write this myself I will need to implement it in both versions (don't want to tease the main Cura community in case this is something others have interest in).

nallath, 2016-03-13:
Pretty much everything in the "new" Cura is a plugin. You can make these plugins as advanced as you want them to be. 

The lulzbot cura edition is indeed forked about a year ago and not compatible with the plugins required for the 2.1 cura.

ianpaschal, 2018-05-24:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

closed 2018-05-24 by ianpaschal