Ultimaker/Cura #662: Download firmware as needed (into TMP)

thopiekar, 2016-03-11:
Related to issue #627, I think about a feature to be able to add a download URL for firmware to the JSON files.

The URL could be even point to a "virtual file", eg.:

```
http://www.printer.vendor.org/download/awesome-printer_latest.hex
```

and Cura will follow according to the http response to

```
http://www.printer.vendor.org/download/awesome-printer_999.hex
```

.. so vendors will be able to provide updates, without modifying the Cura package or waiting for a new release.

Here is a suggestion how the firmware update process could work.

```
1. Check for a local copy of a firmware
2.1 Check for the URL tag in the printer JSON (If there is no tag skip to 4.)
2.2 Check for a available internet connection (If there is no connection skip to 4.)
2.3 Ask the server for the md5sum of the latest firmware, which is online.
2.4 Create a md5sum of the file for comparisons 
3. Download the firmware as needed as tmpfile (which will be removed automaticly when the file gets .close()'d again - so no mess with old firmware)
4. Flash the latest firmware.
```

Finally I would also love to see a function adding the .hex manually. This could be done using two seperate tabs, one for "automatic" and one for "manual" firmware update.

ianpaschal, 2018-02-26:
Closing for ~2 years of inactivity.

closed 2018-02-26 by ianpaschal

Ghostkeeper, 2018-03-16:
The "manual" upload is implemented now. Maybe that can be considered sufficient.