Ultimaker/Cura #663: [2.1] fan speed doesn't seem to change

thopiekar, 2016-03-11:


daid, 2016-03-12:
We've noticed a bug with the fan speed in relation to material selection ourselves. So maybe this is the same bug? (The fan speed in the material selection seems to overrule the use value)

nallath, 2016-03-13:
There is also a reported bug that the fan will always go on, regardless of what you set at "fan full on at height" value.

thopiekar, 2016-03-13:
If you mean the issue report I made before, then it was a firmware bug that
bq already fixed.
Isn't it closed?

nallath, 2016-03-13:
It seems to be an engine issue. It actually sends a g-code that puts the fanspeed at full on.

thopiekar, 2016-03-13:
Ok, in the past I sent the command to change the fan speed manually via
serial console and had the behaviour too. So contacted bq, they confirmed
that and they fixed it in the latest firmware.

nallath, 2017-07-27:
I assume this is no longer an issue, as we had a number of fixes regarding this recently. If this is still an issue, please re-open it.

closed 2017-07-27 by nallath