Ultimaker/Cura #667: [2.1] Cherry-picking commits made on USBPrinting

thopiekar, 2016-03-15:
Yesterday I sent a PR where I translated the if-clauses into dictionaries.
Do you want this in [2.1]? If you are too busy I would try to prepare another PR for [2.1] where I cherry-pick the commits from [master].

https://github.com/Ultimaker/Cura/pull/664

nallath, 2016-03-16:
How big do you think the chances of regression are? That the main concern right now. We should only be fixing bugs in 2.1 (in order to minimise the risk of stuff going wrong)

thopiekar, 2016-03-16:
I think it is save to be moved into 2.1.
It should make easier for vendors adding their firmware into Cura. On top
of the dictionary, which lists the firmware files, there is also a short
description how to add the firmware files, too.

I'm fine if you like to keep it for the next release, but I think it might
be useful for the current release, too.

Ghostkeeper, 2016-03-16:
I agree that it is safe to be moved. The code is fairly simple, really contained and verified by me. But it doesn't add much value either since no change should be visible by the user.

nallath, 2016-03-16:
Then let's just leave it how it is now (and save thopiekar the effort ;))

closed 2016-03-16 by nallath

thopiekar, 2016-03-16:
OK :+1: