Ultimaker/Cura #670: [OS X] Store user settings in standard location

nilrog, 2016-03-17:
In v15.04.3 Cura stored settings etc. in the proper location on OS X (which is **~/Library/Application Support/appname/**).

But the new 2.1 beta doesn't store it's settings there. Instead it uses a hidden directory in the root of the home directory (**~/.cura/**), which is how it usually is on a Linux system.

To further confuse users, this directory does not contain a **"plugins"** subfolder where you can put Cura plugins. But if you manually create that subfolder and put plugins there they are found by Cura. So an empty folder should be put there by default.

nallath, 2016-03-18:
I'm not so sure about your first point (i'm not familliar enough with mac to say anything about "proper" locations), but the second thing is definately something we should look into.

nilrog, 2016-03-18:
https://developer.apple.com/library/mac/documentation/FileManagement/Conceptual/FileSystemProgrammingGuide/FileSystemOverview/FileSystemOverview.html#//apple_ref/doc/uid/TP40010672-CH2-SW1

nilrog, 2016-03-18:
And the fact the the current "stable" version uses that folder.

fieldOfView, 2016-03-18:
As an occasional OSX user I agree with @nilrog

fieldOfView, 2016-04-13:
For our reference: CURA-1390

awhiemstra, 2016-04-13:
Yes to saving in the standard location (though I find the standard location incredibly weird on OSX), no to always creating a plugins folder. The directories are created as needed by the application. Adding logic to create directories when not needed gets ugly way too quickly. The plugin dir should simply be documented a bit better.

nilrog, 2016-04-13:
Well, it is not any weirder than the location in Windows... ;)

And having the folder there would make it a lot easier for user to know where to put them...unless you plan on a fancy UI to handle installing plugins. You know, users tends to skip reading documentation.

Ghostkeeper, 2016-04-22:
Fixed by awhiemstra in https://github.com/Ultimaker/Uranium/commit/9cc143057963937fa17b4c046d33e8b214c11abe.

closed 2016-04-22 by Ghostkeeper