Ultimaker/Cura #671: Questions about RemovableDrivePlugin

thopiekar, 2016-03-18:
I just looked over the code and got different questions:
* What stores self._drives? Is key always a mount point or can it also be a id. The value of the key is likely the label of the drive, isn't it?
* How does the plugin decide which code to take? Eg. LinuxRemovableDrivePlugin.py?

nallath, 2016-03-18:
Check the __init__.py. if you add a plugin the register in __init__.py is called. There is a check for your OS (so based on that one of the plugins is added)

I think that for self._drives the key can be an ID, but i didn't write that plugin, so you'r best off asking @awhiemstra. We should have written more documentation on this though.

thopiekar, 2016-03-18:
Don't ask why I didn't take a look at the ```__init__.py``` file :smile: 

Ok, because I was playing with the LinuxRemovableDrivePlugin.py and for a reason it seems that the variable ```drives``` has as ```key``` the path to the root of the mounted filesystem and ```value``` as label of the drive. I get no errors or warnings and I guess there is a check like ```os.path.isdir(key)``` somewhere in Cura which doesn't throw any error message and silently acts like there was no element in ```drives``` given

closed 2017-07-27 by nallath