Ultimaker/Cura #674: 2.1_3-11: Posted speeds incorrect

CCS86, 2016-03-19:
Hi guys, the new build is awesome!

I set my print speed to 25 and infill speed to 35:

[image]

I could have sworn that the outer walls were running at the same speed as the infill. I looked through the gcode and wasn't able to find that, but I did notice the posted speeds were wrong. Not sure if this is a frontend or engine issue.

Infill is running at 30mm/s:
```
;TYPE:FILL
[UM34_New Handle Boss.txt](https://github.com/Ultimaker/Cura/files/180638/UM34_New.Handle.Boss.txt)

G1 F900.000 E759.21390
G1 F1800.00000 X164.106 Y168.516 E759.22150
G0 F11400.00000 X163.998 Y167.775
G1 F1800.000 X163.246 Y168.528 E759.24162
G0 F11400.00000 X162.386 Y168.539
G1 F1800.000 X1
```

And outer walls at 26.25mm/s
```
;TYPE:WALL-OUTER
G1 F900.000 E757.29135
G1 F1575.00000 X159.237 Y166.450 E757.29724
G1 X159.585 Y166.318 E757.30310
G1 X159.902 Y166.121 E757.30898
G1 X160.174 Y165.867 E757.31485
```

CCS86, 2016-03-19:
Ah ha, the outer was running faster than the inner, but not as fast as the infill:

`;TYPE:WALL-INNER
G1 F1350.000 X165.768 Y161.146 E717.71648
G1 X165.048 Y161.597 E717.72987
G1 X164.417 Y162.184 E717.74344
G1 X163.909 Y162.894 E717.75720
G1 X163.548 Y163.681 E717.77084
G1 X163.472 Y164.017 E717.77626
G1 X163.095 Y164.007 E717.78221
G1 X163.001 Y163.456 E717.79101`

CCS86, 2016-03-19:
I tried posting code for:

Print speed: 1
Infill speed: 2
Shell speed: 3

and this is what I got:
[UM34_New Handle Boss2.txt](https://github.com/Ultimaker/Cura/files/180664/UM34_New.Handle.Boss2.txt)

CCS86, 2016-03-21:
I installed the 3/18 build and tried again.

Settings:

[image]
[image]

Posted Speeds:
W.I.: 270 = 4.5 mm/s
W.O.: 1575 = 26.25 mm/s
Fill: 315 = 5.25 mm/s
Skin: 900 = 15 mm/s
E_r: 1549.27859 = 25.821... mm/s
E_p: 900 = 15 mm/s

I thought maybe the low speeds were causing an issue, so I multiplied them all by 10. Then I  decided to enable all speed related settings to see where the were calculated with these inputs.

[image]

And here were the posted speeds:
W.I.: 270 -> 675 = 11.25 mm/s (2.5x)
W.O.: 1575 -> 1575 = 26.25 mm/s (0x)
Fill: 315 -> 1125 =  18.75 mm/s  (3.571x)
Skin: 900 -> 900 = 15 mm/s (0x)
E_r: 1549.27859 -> 1320 = 22 mm/s (.852x)
E_p: 900 -> 900 = 15 mm/s (0x)

Something clearly seems to be wrong in this calculation; unless I'm missing something bigger.

Thanks,
Lars

BagelOrb, 2016-03-21:
Are you sure the minimal layer time isn't kicking in?
That would cause all speeds to be lower..

CCS86, 2016-03-21:
Yup, those are both set to zero.

BagelOrb, 2016-03-21:
Are you sure the bottom layer speed isn't taking over?
Perhaps you should set the Number of Slower Layers to zero.

Expected behaviour:
from z=0 to layer [nr slowdown layers] the speeds of the different features gradually change from bottom layer speed to the speed settings for the different features (e.g. outer wall, skin, etc.).

CCS86, 2016-03-21:
I was pulling speeds from layer 130, so I can't imagine that is the case.

Are you guys seeing correct posted speeds?

BagelOrb, 2016-03-22:
I am using some regex magic program to get a nice insight into the F-values:
~/bin/substitute.pl y 'while(/(\d+\t(;LAYER:\d+|;TYPE:[A-Z]+\-?[A-Z]*|G1\ [^\n]*F(\d+(\.\d*)?)))/gsm) { print "$1\n"; }' ../test_lines.gcode | uniq -f1 > test_F_values2.txt

This gives me stuff like:
387518	;LAYER:470
387520	;TYPE:FILL
387521	G1 F3150.000
387527	;TYPE:WALL-INNER
387528	G1 F3150.000
387565	;TYPE:WALL-OUTER
387566	G1 F1800.000
387614	G1 F9000.000
387626	;TYPE:FILL
387628	G1 F3150.000
387638	;TYPE:WALL-INNER
387639	G1 F3150.000
387716	;TYPE:WALL-OUTER
387717	G1 F1800.000
387823	;TYPE:SKIN
387824	G1 F1800.000
387830	G1 F5669.291
387831	G1 F9000.00000
387836	G1 F9000.000
387843	G1 F6590.38902
387844	G1 F5950.41322
387854	;TYPE:WALL-INNER
387856	G1 F3150.000
387896	;TYPE:WALL-OUTER
387897	G1 F1800.000
387994	G1 F6666.667

Especially in the skin speeds may go up to 9000 (150mm/s), because small lines get combined into lines with a smaller width than normal, which causes the speeds to be adjusted to keep the flow constant.
The same thing also sometimes occurs in the infill and sometimes the TYPE:SKIN tag is missing.

Everything seems to be working as normal from the engine view of things.

However, there might be a frontend bug which sent the wrong settings.
Please try to reproduce the bug and clearly mention all steps you took.
It would also be nice to have the engine log and the log of the frontend.

BagelOrb, 2016-03-22:
Something is definitely up.
Setting the bottom layer speed to 3000 can cut half of the time off from my 5 hour print!
Somehow the bottom layer speed has more effect than it should.

[Continues debugging]

BagelOrb, 2016-03-22:
Found it, fixed it!
(on 2.1 branch, later to be merged to master..)

Thanks for the detailed bug description!

closed 2016-03-22 by BagelOrb

CCS86, 2016-03-22:
No problem. Glad I could help!