Ultimaker/Cura #675: New Language

CoderSquirrel, 2016-03-21:
HI, i wanna start translating cura to portuguese (pt-br) and i wanna know against which  branch should i made the pr?
And how the best way to make it?
btw i already start to make the translation, isn't complete, not close, yet, but i alread have the cura.po and fdmprinter.json.po files being rewriten in portuguese.

thopiekar, 2016-03-21:
Is your translation working? I just worry that you've used ```pt-br```, because the german translation doesn't use ```de-de```, but ```de```, for example. But I can be wrong :wink: 

Don't know what the Ultimaker people will say in detail about your PR #676, but I guess they will like to see your PR rebased on the [master] branch.

However, you'll get an better answer from them tomorrow and I'm sure your contribution on your language will be welcome :smile: 

Regards

CoderSquirrel, 2016-03-21:
hey @thopiekar i didn't know about this problem :s 
I just put this way because there pt from another countrys too, but i cn easy change for the ptbr with no - ;D

thopiekar, 2016-03-21:
Don't worry. I'm just sure that Cura expects a specific pattern of the country code and I never saw something with a ```-``` in it so far.
Is it working so far for you? It is possible to change to ```Brazil```?

fieldOfView, 2016-03-21:
>against which branch should i made the pr?

Against master. No functional changes will go into 2.1 before it is released.

closed 2017-07-27 by nallath