Ultimaker/Cura #677: Support roof pattern orientation disadvantageous

avogra, 2016-03-23:
Cura version: 1.99.00 on Windows 7 x64
Most of the support roof patterns use diagonal lines. That might be a disadvantage as the top/bottom patterns use diagonal lines too.

Scenario: I have a flat area to be supported and I select 33% roof density. It might happen that the top lines of the roof are in parallel with the bottom lines of that area so that every 3rd line of the part is fully supported by one line of the roof while all other lines are unsupported. Regular support prevents this by using 0°/90° patterns.

If you rotate the roof pattern to 0°/90° orientation, then you would need to rotate the regular support to +-45° to prevent the same problem between regular support and roof

Ghostkeeper, 2016-03-24:
This might mean we'd have to rotate the regular support also by 45°, to reduce overhang.

avogra, 2016-03-24:
Yes that is what i meant with the last paragraph. After reading it again that was a bit unclear.

Do you think this is a valid point at all?

A simple solution could be to add the existing patterns rotated by 45° for support and roof. Then the user is responsible for combining them in a useful way.
Of course it would be more elegant and less error prone, if Cura selected a useful orientation in the first place.

Ghostkeeper, 2016-03-24:
The support (and skin and infill) is oriented diagonally on purpose because an XYZ printer can then use two motors at the same time to move the print head. This can achieve greater speeds with the same accuracy. If we have to make the support horizontal that could increase the print time significantly.

But if we can construe a way to make only the support roof print differently, that would not really be an issue since support roof is a small portion of the total print time (the rest of support is not).

avogra, 2016-03-24:
I think only infill and top/bottom are diagonal at the moment while the support is actually along the X and Y axis. That is what I see in Cura 15.4 and 1.99 at least. So the modification would even help with print speed.

Ghostkeeper, 2016-03-24:
Oh you're right. I'm not level-headed today am I? :confounded:

fieldOfView, 2016-03-24:
/logs issue for auto head leveling

Ghostkeeper, 2016-03-24:
Certainly a good idea to implement then.

BagelOrb, 2017-07-02:
This is only a problem when the Support Top Layers is an odd multiple of the layer height, and the support roof pattern is lines.

ianpaschal, 2018-05-24:
This appears to be fixed. I've tried reducing the support interface thickness incrementally to get both directions of lines and in all cases the actual print surface generates perpendicular lines.

closed 2018-05-24 by ianpaschal

BagelOrb, 2018-05-24:
Hmm. It depends on the support Z gap, though.