Ultimaker/Cura #678: Support Z-distance ignored if smaller than layer height

avogra, 2016-03-23:
Cura version: 1.99.00 on Windows 7 x64
When I set a support z-distance of 0.15mm (default) and a layer height of 0.2mm there is no gap all between support and part. Seems like the z-distance is truncated to units of the layer height. I think it would be more useful to either read the z-distance as minimum value or at least round it to the next multiple of the layer height.
Compare also Cura 15.04.02, which creates a one layer gap between support and part.

Ghostkeeper, 2016-03-24:
So round it up? That sounds like a good idea and should be fairly easy to fix.

avogra, 2016-03-24:
Sounds perfect to me :)

Ghostkeeper, 2016-03-24:
Made an issue for you in our internal bug tracker. I think it could be implemented for version 2.2 or 2.3.

meros, 2016-09-10:
Still a problem? Ran into this today, +1 for rounding up!

Ghostkeeper, 2016-09-11:
No longer a problem. Fixed in 2.3. We're rounding up now.

closed 2016-09-11 by Ghostkeeper

meros, 2016-09-13:
The help text still says rounding down...

Ghostkeeper, 2016-09-14:
Ah :( Good catch. String freeze now though.