Ultimaker/Cura #679: First layer of support gives week adhesion to build plate

avogra, 2016-03-24:
I hope you don't get fed up with me opening one issue after another about support stuff, well anyway... :-]

Im still working with Cura 1.99.00

I used a support with zigzag pattern and 3mm "Line distance" and I'm surprised about first and second layer. The first layer is a grid with lines twice the distance compared to layers 2 and higher. Instead it has some lines in x direction which are not in the higher layers.Therefore every other line 
is effectively a bridge with almost no adhesion to the buildplate. In the actual print, the support quickly went loose. Looks like the first layer is the same as the grid pattern.
The same applies to lines pattern, whereas triangle, concentric and grid had a first layer matching the later layers.

Maybe keep the grid for the first layer but cut the line distance in half, so that every support wall is touching the buildplate?

See screenshots...
Layer 1:
[image]
Layer2:
[image]

Ghostkeeper, 2016-03-25:
Feel free to open up as many issues as you like. It's nice to have a participating community :smiley:

Haven't seen this before. Zigzag used to attach to the bed fine. Maybe @BagelOrb has an explanation?

BagelOrb, 2016-03-26:
Good find!

The problem is essentially that the line distance the frontend sends is not actually the one used in the engine. I'm order to keep the infill density the same a triangular pattern uses thrice and a grid pattern twice the line distance. If the engine wouldn't do that (but the frontend did), it wouldn't change the line distance on the first layer.

This should be an easy fix..

avogra, 2016-03-29:
That means, in 2.1.0, when I use grid and set line distance to 2mm instead of using density, then the actual line distance is 4mm? If got that right, it doesn't sound very logical to me ;-)
+1 for doing this in the frontend :)

Ghostkeeper, 2016-03-30:
Yeah, when using line support the line distance is correct but using grid support the distance is doubled, because it draws additional lines in perpendicular direction and wants to keep the infill density the same. Interesting usability flaw, actually.

BagelOrb, 2016-03-30:
For out reference: CURA-1317

BagelOrb, 2017-07-21:
Fixed a long time ago

closed 2017-07-21 by BagelOrb