Ultimaker/Cura #680: 15.06.03: Cura won't slice model

Mitrasmit, 2016-03-28:
Cura 15.06.03 suddenly won't slice any model. It will load the model, scale it and everything, but does not compute print time, show layer buildup, or allow me to save gcodes. This happens with every model I try (and also with 15.06.01 that I also have installed). The non-beta version of Cura does slice correctly.

I have not made any changes to Cura as far as I know. Am I doing something wrong, or is this caused by the software? 

[image]

Ghostkeeper, 2016-03-29:
Getting any errors?

In any case, there have been enormous improvements to stability since 15.06.03. The beta page on http://software.ultimaker.com/Cura_closed_beta/2.1/ can give you a new download.

Mitrasmit, 2016-03-29:
No errors, it just greys out the Save to Disk-option and doesn't slice. It doesn't give any indication of knowing it's being erroneous.
I don't really understand what to do with the 2.1 download link? 

Anyway, all my beta versions are affected (and use the same settings apparently), so I guess maybe there's a settings file that is corrupted? I am not a programmer, so I have no idea if I'm talking nonsense.

Ghostkeeper, 2016-03-29:
After version 15.06 (which was last June), we continued developing and produced versions 15.10 and then 2.1. They are continuations from the versions you're using and will likely perform better, work nicer, have more features and most importantly, have a lot less bugs like these :v: The confusion, I think, arises from the fact that we switched to a different version numbering system at 2.1, one that matches Ultimaker's 3D printer releases.

As for the issue in version 15.06, your guess is as good as mine. Literally: I'm guessing that some setting got corrupt. I think you're best off downloading a new version from the beta page I linked. Otherwise, you could try deleting your settings folder. The folder is `C:\Users\<username>\AppData\Local\cura`. The AppData folder is hidden by default, but you can type it in the address bar of Explorer if that's easier. The issue is probably in either the `profiles` folder or the `settings` folder, but it's probably easier to delete the entire `cura` folder if you don't mind losing some preferences.

Mitrasmit, 2016-03-31:
Thanks for the replies and your patience. After removing the settings folder, it seems to be working again. I'll give the newer versions a chance as well!

closed 2016-03-31 by Ghostkeeper