Ultimaker/Cura #681: [2.1] Can't reset maximal fan speed

thopiekar, 2016-03-29:
What I get here is a loop of resetting the value and repairing it (exclamation mark).
The console output doesn't tell anything interessant, just that it restarted CuraEngine.

See:
[image]
[image]

fieldOfView, 2016-03-29:
Not sure if I agree with the qualification "bug"; the behavior is correct (though slightly confusing).

Apparently the profile defines an absolute value for Maximum Fan Speed, and Cura rightly shows a notice that the normal calculated value is overridden. After restoring the setting to its calculated value, Cura rightly shows you that the setting has been changed from the absolute value stored in the profile. The confusion here is that the absolute value is the same number as the value that is calculated from the Fan Speed value.

The "bug" is not so much in Cura's logic, but in the profile. Specifically, the Maximum Fan Speed being overridden to 100 in the PLA material profile. However, removing that value from that profile causes the Maximum Fan Speed not to be reset when switching back from ABS as a material.

(Note: this does not happend on UM2 or UM2+ machines because they don't use the generic material profiles; use UMO to reproduce)

Ghostkeeper, 2016-03-29:
I'd still categorise this as a bug though, even if it is a bug that will likely only be solved with the design refactoring we're doing for 2.2!

thopiekar, 2016-09-19:
Doesn't happen anymore.

closed 2016-09-19 by thopiekar