Ultimaker/Cura #684: [2.1] Flickering top layers

thopiekar, 2016-03-30:
They most of all appear when moving the slider fast.
I just made a video of it and hope it wasn't fixed since last weekend when I rebuilt Cura.

https://youtu.be/nMF-HzJyD6w

Ghostkeeper, 2016-03-31:
The video is private :(

thopiekar, 2016-03-31:
Ooops. Now it is "Unlisted".

thopiekar, 2016-03-31:
I guess the only problem here is that the top layer gets generated while sliding, but it should be generated, in my point of view, when the mouse click on the slider gets released.
This will also remove unneeded calculations while choosing the height. If we were faster on doing this, updating the top layer on every change would be fine.

Ghostkeeper, 2016-03-31:
Ah, I see what you mean. It's lagging behind. I've seen that too, and on a few occasions could even get it to remain stuck on a different layer than the one I've selected (because it's essentially a race condition there).

thopiekar, 2016-08-29:
Remember that I played with that on [master] recently.
Afaik it was fixed.

closed 2016-08-29 by thopiekar