Ultimaker/Cura #685: [2.1] Rotate tool: light gray text on white background

thopiekar, 2016-03-30:
I would also love to see a dark blue border around the box. Above the box is inside the model so you can see it, but on the normal white background you won't. So a border would highlight it.

[image]

Ghostkeeper, 2016-03-31:
The text properly appears black for me: http://i.imgur.com/sdxtxCX.png

thopiekar, 2016-03-31:
That's strange. Maybe we are using a system value here. I'll recheck using
a light theme, as I'm using "Breeze Dark" here (KDE).

thopiekar, 2016-03-31:
Yes, it is! It seems to use the default color whereas the rest of the UI uses light colors.

[image]
[image]

Ghostkeeper, 2016-03-31:
Good find!

fieldOfView, 2016-04-04:
Fixed in https://github.com/Ultimaker/Cura/commit/a32c00141ac28ff116ed11eb2b10f77f989cf812

closed 2016-04-04 by fieldOfView

fieldOfView, 2016-04-04:
You're welcome. Taking this a bit off-topic... In some of your previous screenshots, I noticed black text on a dark background that I found quite hard to read (I remember a USB print progress dialog). Is that a combination you would expect with the theme you use? I'm not overly familiar with themes in Linux.

thopiekar, 2016-04-04:
I guess the best text color there, would be the default given by Qt5 (theme
used by the desktop).
In that case it should be the light gray color that was found in this issue.

So I would be fine with that or to overwrite all colors. But bad thing
would be that the window decorator will still remain black in my case.

fieldOfView, 2016-04-07:
@thopiekar I think I fixed most of the occurrences of the text in dark color on dark background with your theme. Could you check?

thopiekar, 2016-04-07:
@fieldOfView, of course! I'll give feedback as soon as possible :wink:

thopiekar, 2016-04-07:
Sorry, but the color is still not black.

thopiekar, 2016-04-08:
@Ghostkeeper @fieldOfView Better reopen this issue so you don't forget to fix it before releasing 2.1 :wink:

fieldOfView, 2016-04-08:
It won't be fixed for 2.1, but it has been fixed in master. Even though it is only a "cosmetic" change with little to no risk, we are too close to release to merge this in.

thopiekar, 2016-04-08:
Ah, ok. I thought you wanted to fix that in 2.1, too. I've got a PPA for
master builds, but don't use it at the moment.