Ultimaker/Cura #687: CPackConfig.cmake is regenerated when building

onitake, 2016-04-05:
cmake seems to regenerate CPackConfig.cmake when Cura is built and installed, with a lot of local information. Git will then report the file as "modified", which could lead to accidental commits.

https://cmake.org/Wiki/CMake:Packaging_With_CPack#Using_CMake_variables_to_configure_CPack suggests that it's better to put all the non-local settings into CMakeLists.txt instead, or a separate file that is not regenerated. Then, CPackConfig.cmake could be removed from the repository and added to gitignore.

awhiemstra, 2016-04-06:
CMake should not touch the CPackConfig.cmake in the source dir if you are running a build from a separate build directory. Running a build from within the source directory is not supported by us. That said, the naming of the file here is wrong as it mostly just sets up the variables for CPack and at the time I created that file I did not know CPack generated a file with the exact same name.

closed 2017-07-21 by BagelOrb

Ghostkeeper, 2017-07-30:
Problem still exists, but only in Uranium.