Ultimaker/Cura #688: Bridge intitial infill should span abutments

gcanzalone, 2016-04-05:
The direction of the initial infill layer for bridges should span abutments instead of attempting to connect perimeters as the current version does. Bridges as currently implemented sag badly as a result of connecting infill to only thin perimeters.

BagelOrb, 2016-04-06:
The size of the abutment is not related to the direction of infill layers. As it is now it's simply the breadth of the walls.

Bridging is currently in quite a poor state. It will be renovated in hopefully not too much time.

carbonrim, 2016-06-07:
BagelOrb would you mind to answer what does it mean "not too much time"? Is it 2 months or a year? I'm just curious, because I really like Cura and use it, but only for projects/objects not demanding bridges (I don't like supports).

BagelOrb, 2017-07-21:
It has been too much 😓 
Again I'm inclined to say it will be here in a couple of months, sooooo maybe at least another year.

:(

ianpaschal, 2018-03-26:
This has been fixed.

closed 2018-03-26 by ianpaschal