Ultimaker/Cura #690: Auto rotate feature

iHateInventNames, 2016-04-06:
It would be nice to have auto-rotate feature like Repsnapper has.
Good model to test [peppa](http://www.thingiverse.com/thing:410898/#files).

nallath, 2016-04-06:
Do you mean automatically calculate the best printing orientation?

iHateInventNames, 2016-04-06:
Yes if printing orientation means select all needed rotation to get best cost effectiveness or quality

Ghostkeeper, 2016-04-07:
There is the Lay Flat feature but it only really accounts for overhang.

mokusolo, 2017-01-12:
Curious about the problem that the model moved up and down after rotation in 2.1(I know you've fixed it in 2.3), I found RotateTool.py where the function layFlat() is,  but can't figure out how it's invoked after the MouseReleaseEvent. I think the jumping is caused by the layFlat, and I want to close this feature. Most time it's good and expected, but always with large and complex files I don't want that, and I can replace it with "center model on platform".
 Could you please give me some advice? Thank you.

awhiemstra, 2017-01-12:
The push up/down is not handled by the tool, it is handled by the platform physics. See https://github.com/Ultimaker/Cura/blob/master/cura/PlatformPhysics.py#L85 and further.

iHateInventNames, 2017-01-12:
Yes @mokusolo, this algorithm is implemented in open source Repsnapper, you can find it in its code.

mokusolo, 2017-01-12:
Thank you so much. My misunderstanding...

nallath, 2017-07-27:
Calculating the best orientation is now handled by a seperate plugin.

closed 2017-07-27 by nallath