Ultimaker/Cura #691: Memory leak on fresh install and load

rphauck, 2016-04-07:
Running Windows 10 64-bit install. After clean installation and starting the software, without selecting a machine or doing anything, the software will consume all memory on the machine and stall Windows. Cura 3/18 build.

nickthetait, 2016-04-08:
I believe @kakaroto saw this yesterday using the 3/18 build too.

fieldOfView, 2016-04-08:
This is but a hunch, but could you remove the Doodle3D folder in the plugins folder, and then check again?
The Doodle3D plugin folder is added to the package when Cura is built, and most of us don't test with the plugin installed. Since we don't see the memory leak during development, that could just be it.

rphauck, 2016-04-08:
Removed the Doodle3D folder and same situation persists. As soon as I select "Finish" with the default Ultimaker 2 machine selected, the memory leak stops and doesn't repeat again until I delete the "C:\Users\XXXX\AppData\Local\cura" folder (where XXXX is my user name).

fieldOfView, 2016-04-09:
Thanks for testing this. I have reproduced the leak on Windows and Linux. We will look into this.

nallath, 2016-04-12:
We just managed to fix the issue without removing the doodle plugin. See https://github.com/Ultimaker/Cura/commit/cdb235740d52045accff835f01c7f6c021046270

closed 2016-04-12 by nallath