Ultimaker/Cura #693: Newly loaded items not focused for operations

Gareee, 2016-04-11:
I just noticed when loading a new object, the item isn't in focus, so I can't just select scale/rotate/ect. I'd think it would be an easy thing to have cura auto select a newly loaded object.

Ghostkeeper, 2016-04-12:
Agreed. This would be useful and easy to implement too.

Ghostkeeper, 2017-01-31:
This was just rejected by our change control board for being unimportant.

closed 2017-01-31 by Ghostkeeper

MarcusWolschon, 2017-01-31:
Why are unimportant things rejected instead of prioretized to the bottom of the backlog?

nallath, 2017-02-01:
Because there are over 800+ issues, most of them are very old. We can leave them in, but it makes it freakishly hard to keep track of them and pretty much any issue will have priority (meaning that it's extremely unlikely that it gets done).

I don't like it either, but right now the overhead is killing us. It's costing me hours a day, which are better spent on actually making stuff.

fieldOfView, 2017-02-01:
I think it is important to note that there are github issues (currently 317 open) and internal "JIRA" issues (of which there are apparently ~800 open at the moment).

All said and done though, there should be room for features that UM is not going to assign a high enough priority to, to be picked up by certain community members.

thopiekar, 2017-02-01:
Maybe instead of "killing" all ideas and issues inside JIRA, we should move these issues into GitHub?
The benefit would be that others (e.g. community) can jump in and work on these, if they are interested?

I think that better than just closing tickets and forget about ideas and thoughts...

nallath, 2017-02-01:
Good points. The whole github & jira connection is not really what it should be though. It's a bit of a hassle to have two systems side by side where some of the issues are linked and others are not. 

Any ideas how to best do this? I'm willing to champion any suggestions you guys have to within UM. Come think of it, it's also something that affects aleph objects as well. @nickthetait What is your take on all this?

nickthetait, 2017-02-02:
Just a heads up that I recently stopped working for Aleph Objects. I'll remain involved with Cura, but won't be able to spend as much time on it.

Anyways, I definitely understand how much effort goes into managing a bug tracker system. Seems to me like consolidating on a single tool would be the most valuable. Making the Cura project on Jira publicly visible seems the most obvious. It should be possible (at least Phabricator can) to keep certain tasks secret, if there actually is a need.

Afterwards, you can setup a "community only" column. This would contain tasks which Ultimaker does not plan to invest time in, but would probably merge.

Provides a number of benefits:
* Community knows which tasks aren't being worked on
* Start shrinking your backlog. Less intimidating to your development team :wink: 
* Cut waste of managing duplicate issues
* Help reduce duplication of effort. This is a small example https://code.alephobjects.com/T490 vs  https://github.com/Ultimaker/Cura/pull/1078

The messier option would be a tool to help bridge the gap between the two systems. Something like this https://github.com/integrations/jira-software

fieldOfView, 2017-02-02:
I'm sorry to read that Nick. On the other hand it is great that Cura is gaining another independent contributor!