Ultimaker/Cura #694: Duplicated item not showing in modified rotation

Gareee, 2016-04-11:
This is kind of an odd one. I loaded an object, and it loaded backside up. I then rotated it for printing on its back. I duplicated it, and the duplicate showed that it was not rotated on its back, but in it's original loaded orientation. I did this twice more with the exact same results for a total of 3 copies of the original object.

I then rotated them all the copies onto their backs so they would print like the first one, and then saved to sd card and printed. The only one that printed right was the original object, and the rest all failed, trying to print upside down!

The only thing I can think of is the copy preview is just showing the original object preview, not its actual rotated state, and when I "corrected" them, I was actually flipping them all back to the original object's load state, which was with the bottom facing up.

nallath, 2016-04-12:
With what version did you test this?

Gareee, 2016-04-12:
This should be in the latest version... I think 318. I'll verify that as soon as I get to my surface and confirm.

Gareee, 2016-04-12:
[image]
The upper object is the original, which has been rotated to lay on its flat back. the bottom object is its duplicate, and you an see the post in the middle is not facing down. It looks like that object needs to be rotated like the original one was, but its actually already rotated, but it's preview isn't reflecting that change of rotation.

nallath, 2016-04-12:
The preview is actually the way we send it to the engine. Well, we're going to release a new version tomorrow, so could you check if it works there?

Gareee, 2016-04-12:
Will do, and I'll report back here if the issue is resolved or not.

Gareee, 2016-04-13:
Ok, I just checked the new version out... if I duplicate an object that has been rotated, should the new object adopt the rotation from the original, so they are oriented the same in the preview?

If they are, then the bug still exists.

The bottom most object in this shot is the original, and the duplicates are above it. I can't attempt printing this to see if they are really rotated but have an incorrect preview display until another print finishes. 
[image]

nallath, 2016-04-14:
Copied objects should be exact duplicates of the object (including rotation)

Gareee, 2016-04-14:
I'll try with a different object. Maybe it's loading something that's so small scale? I can attach the object if that's possible here for you to test with.

nallath, 2016-04-14:
I very much doubt that it has anything to do with small scale. I tried with multiple models, both big & small.

Gareee, 2016-04-14:
I'll try a uninstall reinstall and see if that cures the issue. I just tried it with a larger object, and the exact same thing happened.

Gareee, 2016-04-14:
Ok, I recorded a session with mixed results.. some objects duplicated as expected, others did not. I'll post the session recording once its done uploading, but in the mean time here is one of the files that had the flip issue.. can someone try this on their install and see if the problem occurs? Its happened here 100% of the time on this file.

Gareee, 2016-04-14:
For some reason it looks like I'm unable to attach a small zip file.. heres a dropbox link with that obj in a zip archive.

https://www.dropbox.com/s/kimji1tsklb6esq/eye.base.ZIP?dl=0

Gareee, 2016-04-14:
Here's the video link:

https://youtu.be/baqxu0N0150

Ghostkeeper, 2016-04-15:
I've tried that with the eye base and it indeed reproduces. It's also wrong in Layer View.

Ghostkeeper, 2016-04-15:
The small scale shouldn't matter at all unless the issue is somehow linked to lay flat, which doesn't seem to be the case here.

nallath, 2016-04-15:
It's in our internal jira tracker, so we will have a look at it. Thanks for going through all that effort to make movies. You really make our job a whole lot easier :)

Gareee, 2016-04-15:
I kind of needed to.learn to do it anyway. On the Microsoft surface 4 pro, you just launch the Windows 10 Xbox App (I think it works on all Windows 10 systems) and then while in the app you want to record, hit the winkey+g and it pops up a record panel, and you just hit the record button. When you want to turn recording off, you again hit winkey+g and hit the stop record button.

You can the edit or crop the video in the Xbox game app, and resave it. VERY easy!

I googled this yesterday, and it's nice to have a preinstalled app for screen recording. This could be used for some quickie tutorials you could make available on the ultimaker website for new users.

nallath, 2016-04-15:
We're thinking about doing that, but it will probably take quite a bit of time before we do it, as we just have too much to do.
If you're interested in doing it, I think there is a pretty big audience for these kind of things.

Gareee, 2016-04-15:
If I really knew what I was doing as a pro, I'd consider it, but odds are I'd just do the worng stuff and confuse people even more.. LOL!

nallath, 2016-07-08:
So this issue finally manged to pop up on our internal tracker again. Even though we did quite a bit of refactoring since then, it still occurs.
Oddly enough, your model is the _only_ one so far that exibits this behavior.

nallath, 2016-07-08:
And as the nice sneaky bug that it is, I only managed to reproduce it once.

Gareee, 2016-07-08:
Sadly, I have to say I use the old version 90% of the time now because I just have too many oddball issues with the new cura. The weird scale display, and not being able to multiply objects just drives me crazy. The scale issue is the biggie for me though...if I want something scaled up 200%, I need for all the axis scales to display 200%. I scale things almost 100% of the time.

nallath, 2016-08-17:
I manged to fix it! Rejoice.

closed 2016-08-17 by nallath

nickthetait, 2016-08-17:
That is one persnickety bug you just squished there. Nice work!