Ultimaker/Cura #695: Feature enhancement request

Gareee, 2016-04-12:
Can the colors of the various axis when displayed always be done in the proper color matching the preview display for consistancy? Text entry boxes always use black, and in the preview for rotation, instead of the usual red/green/blue one of the overlay circle controls is yellow. (And typically for the colorblind yellow is very hard to see on a gray background.)

fieldOfView, 2016-04-12:
Sorry, I have read this a couple of times, and I don't understand what you are trying to say. What yellow overlay circle controls are you refering to? Could you perhaps add a screenshot or a mockup image to make it clearer?

Gareee, 2016-04-12:
I actually JUST noticed you HAD added colors for the X/Y/Z field in the scale pane.. but only on the axis descriptor. Maybe the text entry field boxes could be the same color, or they could be in bold to make them stand out more so idiots like me would see them better? LOL!

(I'm finding I'm working faster in the older cura versions for whatever reason, and it doesn't have the color axes display the new version has.. its all black, and I'm bouncing back and forth between versions, so probably mixed that up in my head.)

Why I thought the green axis rotate control was yellow was beyond me. Its color coded as it should be.

I need to try to work less with a flu. My apologies for the confusion.

fieldOfView, 2016-04-12:
No problem, it has happened to most of us. Thanks for your suggestions!

closed 2016-04-12 by fieldOfView

Ghostkeeper, 2016-04-13:
Maybe also make the borders of the text boxes colour-coded?