Ultimaker/Cura #696: Feature enhancement request: All extended options window

Gareee, 2016-04-12:
While I like being able to selectively add to the quick access scrollable pane, I really find I miss being able to just open the overlay window with all the extended advanced options available.  Is that still available, (and i've missed where it is located) and if not, can it be added? I really liked being able to look over all options at once, and with the scrollable pane, you can't do that because everything isn't on screen at once.

fieldOfView, 2016-04-12:
Your request is noted, and you are not the first to request something like this. I am a bit hesitant to create a third way to edit the same settings (simple mode, advanced mode, and an "all options window"). Note that we now have around 200 options, and there is no way you can look over all options at once. 

Personally I would rather improve the advanced mode to a point where it is at least as usable as what you suggest, for example by allowing a second column if the window is wide enough and by making it easier to discover/find settings and showing them in the sidebar.

Gareee, 2016-04-12:
I can understand your hesitancy for redunant display options. There were actually the three modes in the older versions of cura though... simple, advanced, and expert. Maybe the old expert pane could just be reutilized, but have the same display options similar to the new advanced pane? (Or maybe have tabs for the related operations?) The issue from a user standpoint, is bouncing back and forth a lot more in the new cura in the old one.

There was talk of possible theme options that might help on the current advanced pane. If Print Job and setup could be collapsed, that would free up 1/3rd of my right column.

I'm finding I usually leave my most used panes open in advancced... which means I'm finding I'm scrolling up and down looking for a setting I need to change, and then going back into preferences, to make something visiable I can't locate in advanced, and then AGAIN scrolling up and down in the new advanced pane trying to locate it to actually set it.

I can't honestly think of any use at all for the simple mode in the new cura. For a brand new user, maybe for the first two weeks, yes.. but after that, they are going to run into various print issues, and have to learn how to use at least some of the advanced settings anyway.

I'm actually finding when I want to get something done efficiently, I'm going back to 15.04.5 because I don't want to fiddle around hunting for things in the new interface. I can get to what I want faster in the old interface with a single shortcut keypress to the expert settings.

I'm also a fairly new user to cura, only since Jan. Changing my behaviour should be easy because I'm still learning it and am not really set in my ways yet, but I'd think legacy users who are more resistant to change might be more displeased and vocal in the changes in the new interface.

From a support standpoint, you also want to get all users onboard with the newest version for troubleshooting their issues, instead of trying to figure out which version they are using, and trying to get them to use the updated one instead for whatever reason they chose to stick with the older version.

At any rate, hopefully my rambling is food for thought. ;)

Ghostkeeper, 2016-04-12:
We've had good feedback on the simple mode so far. I agree that it could use some extension though as it is currently a bit too simple perhaps. For one, we could probably add a speed slider there (that only adjusts speed, not the speed/quality slider that was the original idea since that requires extensive profile testing).

As for advanced/expert mode, the original idea was that the visibility panel is the new "expert" mode. I think it's working fine but realise that I'm looking through the developer's goggles.

I have about 1,5 years of printing experience, albeit sparse, and know exactly what each of the settings do. Still I find that for nearly all of my prints I'll just have to adjust a few: Profile, infill density, wall count, adhesion type, print sequence, maximum retraction count, and a few others. So I just show those and hide all the rest.

Maybe someone less versed in Cura's options would have to open the setting visibility menu much more oftener when they search for an option that they aren't sure exists. Like "is there a way to leave some more space between walls" or "can I make it never do a retract when it's going to inner polygons". If that's the case, maybe we can find a way to make this searching easier.

What if we add a search-box to the advanced panel that *searches in not-yet-displayed settings*? I'm imagining it a bit like how you add recipients to an e-mail. You type something and suggestions appear, and when you click on one of those or press enter, the selected setting gets made visible and the focus is placed immediately on the setting.

nallath, 2016-04-13:
I think that a filtering, much like we do in the visibility tab, would be amazing.

fieldOfView, 2016-04-13:
A way to make advanced mode more accessible to power-users who have selected a large number of settings to be visible would be to add a search-as-you-type filter to the sidebar.

Taken a step further, this could replace the Setting Visibility pane of the preferences window, if the filter would also show "invisible" settings (greyed out). A tooltip could include the reason why this setting is not currently visible.

Users would benefit by typing eg "temp" and seeing a list of all temperature settings, or typing "supp" and seeing all support-related settings, right in the sidebar.
(CURA-918)
This would make a trip to the setting visibility page unnecessary.

Gareee, 2016-04-13:
Part of my issues might be because I am a modeller, not just someone who downloads and prints models others have created. There are many options in printing I don't understand yet, but I'm learning them as needed.

As a result, I'm tweaking things like wall thickness, top layer number and such to optimize printing newly created models to get the best end result.

It seems like the most common use of 3d printers is "download and print something someone else made", and the simple mode might be best for those people, but other than my first few test prints to learn, almost everything else I'm printing is a new creation, and there are times I need to adjust either the model, or the settings to get the end results I'm shooting for.

nallath, 2016-04-14:
The "I want to tweak everyting to get the best print possible" flow is what the usage we envisioned of the advanced mode.

ianpaschal, 2018-05-24:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

In addition, it's worth noting that there are now setting visibility presets which can be set to show ALL settings.

closed 2018-05-24 by ianpaschal