Ultimaker/Cura #697: Feature enhancement request: Drop down window for object selection

Gareee, 2016-04-12:
Loading an object that scaled properly makes selecting them fairly easy,l, but if you load something scaled way too small, if becomes a real pain to select them. You have to scroll in as far as you can, and fiddle with the preview rotation to see if you can get the object into view to select it. If there was a drop down object selector, you could just select the obj by its loaded name. This would also come in handy for objects you scale up, which then push other objects way off the print bed space, that you need to relocate. Once selected in the dropdown, objects should not only have focus, but the preview display should also auto center on them as well for faster manipulation back to the print bed.

fieldOfView, 2016-04-12:
An object list / "scene graph" could also help for selecting objects once they are grouped, so their Per Object Settings can be adjusted.

Ghostkeeper, 2016-04-12:
Hehe, please warn me before you tell Jaime that we're implementing that (if we are). I'd like to be outside of the blast zone before then.

nallath, 2016-04-13:
Ugh. Meshlist panel. My years old Nemesis. Are you now trying to infect Cura as well? ;)

fieldOfView, 2016-04-13:
If it improves UX, yes. 
Sounds like I will have to hear you out on the faults (and merits) of meshlists over a couple of beers (or cookies)

nallath, 2016-04-13:
I love it when it works. It's just very difficult to get all the behaviours right.

Gareee, 2016-04-13:
A panel was just my first thought.. there could easily be a menu option listing the loaded objects for selection. Or maybe even a icon based drop down list at the top of the translation overlay controls to select the loaded objects that way before using rotate/scale, ect.

Also a question that just sprung to mind.. in the older cura, you can make a selectable number of duplicates.. in the new version, you can only make one at a time. Will the multiple duplicates option be returning in a new build?

awhiemstra, 2016-04-13:
The multiple duplicate option will most likely not be returning as such. Instead we will be making sure duplicate works on multiple objects at the same time and we want to introduce copy-paste functionality.

Gareee, 2016-04-13:
Copy and paste might do the trick then, since you could repeatedly paste the same object down. The reason I asked, was I created some game tokens (Like 1 1/4" discs) and in the old cura I was printing 18 of them at a time. I loaded on, replicated it 17 times, and was pretty much done. Thats obviously a lot longer process in the new version.

I also recorded a video to show the issue I'm having in some cases selecting a loaded object, and why I'd like to either see them automatically focused upon loading, or to be able to select them from a list of loaded objects somewhere. 

This was modelled in Lightwave 3d, using its default world scaling. In lightwave, it measures at 1" across. In cura, it loads at maybe 1/100th of a mm! I've beeen hesitant to change its world scaling because of projects with multiple parts already output with its current scaling.

Please forgive my sinus infection voice.

https://youtu.be/fHxe0NLOSRM

fieldOfView, 2016-04-13:
Perhaps we should have an option to automatically enlarge objects that are unrealistically small, like we have the option to automatically scale down objects that are too large.

Gareee, 2016-04-13:
That would work in this instance, but when you load an object that pushes another off the print space, you still need to rotate all over creation trying to figure out where it went.

I've always been puzzled by the large gray spaces to either side of some items preventing them from being moved closer together, when clearly based on the printed item itself, they could be moved much closer together if those big gray spaces were turned off or reduced in size.

Also, field of view, please reread my original post with the video, and I added some additional observation information.

fieldOfView, 2016-04-13:
The large gray spaces around objects are because you have set the "Print sequence" to "One at a time"; there needs to be space around the object so the print head does not hit the object when it starts printing the next object. If you reset the print sequence to "All at once" you can place objects much closer to eachother.

Gareee, 2016-04-13:
I completely understand that issue, but I'm printing very short items that wouldn't be hit t all by the print head. I'm using some filament that has unsolvable really bad stringing issues, and if I print all at once, it looks like spiderman was weaving them in the printer. Maybe the calculation of required needed space is off a bit?

Ghostkeeper, 2016-04-14:
How about a drag-select that drags a selection rectangle, like in a file browser? Though that would be fairly hard to implement with the current selection system.

Cura is currently not able to detect when an object is small enough (less than the nozzle height) to reduce the disallowed area in one-at-a-time mode. There has been some work on a feature called "towering" that could make this work on the engine side though. It is meant to bunch up parts of layers that are above each other to reduce the amount of retractions and travel moves necessary.

Gareee, 2016-04-14:
Hmmm... towering sounds pretty interesting!

nickthetait, 2016-04-14:
Where is the towering feature work happening? I can't tell from the branches in CuraEngine if there is one for this. Or is it somewhere on Master?

Ghostkeeper, 2016-04-15:
It's a little side project (out of work hours) of @BagelOrb and someone from a makerspace in Utrecht. I don't think they have their source public actually, because I can't find it in CuraEngine's branches either nor in the projects BagelOrb contributed to.

It's just a bit of experimentation and I imagine BagelOrb and Themba wouldn't like me talking about it too much, with expectation management and all...

nickthetait, 2016-04-15:
I see, well it sounds mighty useful :smile:

awhiemstra, 2016-04-18:
> How about a drag-select that drags a selection rectangle, like in a file browser? Though 
> that would be fairly hard to implement with the current selection system.

Actually, this would be fairly simple, assuming we can fork certain things to native code. Since we already render selectable objects to a texture, we can simply take an area of the texture, figure out which objects are in that area and select them. Doing that in python would probably be ~10 lines of code. Getting it to perform well is the tricky bit, since you do not want to do the looping in python.

fieldOfView, 2016-04-18:
We have to make a choice though, if we want to make selection easier with drag-select, or if we want to make rotating the camera easier for touch-based displays. My finger has no right mousebutton!

awhiemstra, 2016-04-18:
That's pretty simple: we can detect if we are receiving touch events or not. When dealing with touch we should implement gestures to take care of some of the camera motion, specifically using two fingers to zoom, pan and possibly rotate the camera. One finger can then act the same as left mouse button, with single tap for selection and tap-and-drag for box selection.

Gareee, 2016-04-18:
As a surface 4 pro user, having touchscreen control is pretty high up on my want list. I'd be fine with a dropdown or list for object selection, if it meant also gaining much needed features.

MarcusWolschon, 2016-05-01:
this would be great to select objects scaled to meters instead of millimeters as they are otherwise too tiny to even be clearly visible at the default zoom level.

nallath, 2016-05-02:
@MarcusWolschon We thought about this, but most models are in mm. We opted against adding a extra question at loading for the few users that need this (eg; Don't impact 95% of the user for the sake of the 5%).
We could do something with auto up-scale though (same at what it does when a model is waaay to big atm)

fieldOfView, 2016-05-04:
See the above two PRs for an (optional, enabled by default) automatic scale operation on meshes that are so tiny that they would be unselectable in the UI. It currently scales up in factors of 10 (specifically for the case of mm vs cm vc m)

ChrisTerBeke, 2017-11-14:
@jackha is working on this as part of his multi-buildplate feature. Closing this issue for now as it's super old and will be fixed soon.

Ghostkeeper, 2017-11-22:
It ain't fixed yet though and it is demonstrably relevant.

closed 2018-11-26 by Appesteijn