Ultimaker/Cura #702: Cura 15.04.05 does not like the new NumPy casting rule

treepleks, 2016-04-17:
Recent versions of NumPy use casting rule 'same_kind' and Cura does not like it:

Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/share/cura/Cura/util/sliceEngine.py", line 346, in _runEngine
    pos += (objMin + objMax) / 2.0 * 1000
TypeError: Cannot cast ufunc add output from dtype('float64') to dtype('int64') with casting rule 'same_kind'

This is with numpy 1:1.10.4-2ubuntu2

thopiekar, 2016-04-17:
How did you install Cura? And please give details on your system. I can just guess you are using Linux.

daid, 2016-04-17:
Would be nice if you could complain at Numpy that they broke compatibility. As I understand why they added the error, but adding it as a default rule in a minor revision is a bit harsh on the rest of the world that just want working applications.

treepleks, 2016-04-17:
Sorry Thomas,

I thought the version and the bug (plus the likely source of it) was enough.

I downloaded the debian file for the Cura web site. I used th last
stable one (15.04.5 64bits).
Cura was installed on an Ubuntu box with version 1.10 of NumPy installed.
Cura starts fine but when an STL is loaded, slicing fails ith the
message I gave in the issue

After digging on why the python code could break, I found that Numpy had
changed the default casting rules in 1.10:

See http://docs.scipy.org/doc/numpy-dev/release.html

            Default casting rule change

    Default casting for inplace operations has changed to 'same_kind'.
    For instance, if n is an array of integers, and f is an array of
    floats, then n += f will result in a TypeError, whereas in previous
    Numpy versions the floats would be silently cast to ints. In the
    unlikely case that the example code is not an actual bug, it can be
    updated in a backward compatible way by rewriting it
    as np.add(n, f, out=n,casting='unsafe'). The old 'unsafe' default
    has been deprecated since Numpy 1.7.

I assume this is the source of the problem since a cast from float to
int is precisely mentioned in the error message.

Thomas

treepleks, 2016-04-17:
Sorry guys... I cannot help here :-(

thopiekar, 2016-04-17:
@treepleks No problem. As Cura 2.1 is going to be released soon, I don't think that this issue will be fixed. You have two options to jump into a more recent version:
1. Use an updated beta package from https://ultimaker.com/en/products/cura-software/list (officially supported by Ultimaker)
2. Or as you use Ubuntu, use my PPA at https://launchpad.net/~thopiekar/+archive/ubuntu/cura. Don't forget to remove and purge the old Cura package out of your system.

gazhay, 2016-05-03:
Confirmed that this is my issue from here https://github.com/Ultimaker/Cura/issues/718

Sadly, cannot upgrade cura as I have a delta printer that newer versions won't let me setup.

gazhay, 2016-05-03:
I have isolated this to one line of code, and you can overwrite the line with numpy unsafe casting

In the file Cura/util/sliceEngine.py line 346

change from 
`pos += (objMin + objMax) / 2.0 * 1000`

to
`pos = numpy.add( pos, (objMin + objMax) / 2.0 * 1000, out=pos, casting='unsafe')`

fieldOfView, 2016-05-03:
Note that for anything 15.04.x-related, the proper place to discuss is here:
https://github.com/daid/Cura/issues

daid, 2016-05-03:
Most likely the easier fix is replacing the "1000" with "1000.0"

ambrop72, 2016-05-07:
Changing to 1000.0 does not fix the problem.
```
    pos += (objMin + objMax) / 2.0 * 1000.0
TypeError: Cannot cast ufunc add output from dtype('float64') to dtype('int64') with casting rule 'same_kind'
```

theoo, 2016-05-07:
@gazhay answer works like a charm on a brand new Ubuntu 16.04 (not an upgrade, I was suspecting the upgrade in a first place).
I also have a Delta printer, so 2.1 is not an option.
It would be nice to have it in the next release. Thanks you for all the work btw.

theoo, 2016-05-07:
Thanks to @gazhay, the solution it working on old (14.04) and recent (16.04) Ubuntus.
I opened a pull request: https://github.com/daid/Cura/issues/1461

fieldOfView, 2016-05-09:
Please continue this discussion here: https://github.com/daid/Cura/issues/1461

closed 2016-05-09 by fieldOfView