Ultimaker/Cura #704: RuntimeError

danilius, 2016-04-18:
Got this error which prevents Cura from starting:

Traceback (most recent call last):
  File "/opt/cura/bin/cura", line 19, in <module>
    import Arcus
RuntimeError: the sip module implements API v11.0 to v11.1 but the Arcus module requires API v11.2

nallath, 2016-04-18:
Oh this one again. Did you do a clean install of cura?

danilius, 2016-04-18:
I installed the deb. I did download the source code and managed to compile it once or twice and then some bugs crept in and I gave up the ghost. So, how do I clean up the mess? That will probably cure this.

nallath, 2016-04-18:
Someone else did report this isse on the cura-build (https://github.com/Ultimaker/cura-build/issues/40). Maybe @awhiemstra @thopiekar or @Ghostkeeper know a bit more about this (I seem to recall that we fixed it, but can't find anything about this :()

thopiekar, 2016-04-18:
I guess here is something messed up. Can you give exact info about what you've done? E.g:
1. Was an older version of cura installed as deb before?
2. Have you tried to install one of the beta packages _after_ that?
3. What is the name of your own built package (filename)?

Additionally, please give detailed info about your installation. I can just guess you are using Linux and a APT-based distribution.

Thanks!

danilius, 2016-04-18:
1) Yes
2) Yes
3) Cura-2.1.0-Linux.deb which I downloaded from here: http://software.ultimaker.com/Cura_open_beta/

I reinstalled the 15.06 version since I need that for work.

thopiekar, 2016-04-18:
And the old version is working fine?

danilius, 2016-04-18:
Yup.

awhiemstra, 2016-04-18:
It is working because that did not use SIP generated bindings. We had this issue before, for some reason either Python or Sip is ignoring all the paths we set up to run Cura and uses a system provided version. @danilius Do you have python3-sip or a similar package installed?

danilius, 2016-04-18:
Nope, not to my knowledge. But I can install it if you think it will make a difference.

awhiemstra, 2016-04-18:
Well no, I expect it to be the cause of this issue. What seems to happen is that python or sip decides it should use the system version and that conflicts with the sip used to compile Arcus. Unfortunately I do not know of a workaround at this moment.

danilius, 2016-04-18:
Oh well.

thopiekar, 2016-04-18:
Ok, please do the following (I'm sorry that I can't test it here locally).

1. Please reinstall Cura 2.1.0
2. Use a terminal and type in ```ls -la ./opt/cura/lib/python3/dist-packages/sip.*```
3. If it only says ```/opt/cura/lib/python3/dist-packages/sip.so``` (like expected), do:
4. ```sudo ln -s /opt/cura/lib/python3/dist-packages/sip.so /opt/cura/lib/python3/dist-packages/sip..cpython-34d-x86_64-linux-gnu.so``` and ```sudo ln -s /opt/cura/lib/python3/dist-packages/sip.so /opt/cura/lib/python3/dist-packages/sip..cpython-34dm-x86_64-linux-gnu.so```

I know that binary modules in Ubuntu use a non-default way to call their binary modules. This will provide the files by symbolic links.

danilius, 2016-04-18:
Did that. When I do ls -lh in /opt/cura/lib/python3/dist-packages I get:

lrwxrwxrwx  1 root root   42 Apr 18 14:43 sip.cpython-34dm-x86_64-linux-gnu.so -> /opt/cura/lib/python3/dist-packages/sip.so
lrwxrwxrwx  1 root root   42 Apr 18 14:43 sip.cpython-34d-x86_64-linux-gnu.so -> /opt/cura/lib/python3/dist-packages/sip.so

Is this correct? Didn't cure the problem mind you.

danilius, 2016-04-18:
I just re-installed the old one and it works fine.

thopiekar, 2016-04-18:
Sorry, for my late reply. Yes, that should be fine. Was it working after that? Did you get a different error?

danilius, 2016-04-18:
Nope, I got the same error. So I reinstalled the old version and that works fine.

thopiekar, 2016-04-18:
Ok, I'm sorry for that.

If you still want to use Cura on Ubuntu you can take a look at: https://launchpad.net/~thopiekar/+archive/ubuntu/cura
Just mention that you are using this PPA if you find any problems and don't forget to remove the currently installed ```cura``` package, eg. by ```sudo apt-get purge cura```.

Regards

danilius, 2016-04-18:
No need to apologise, I'm pretty sure my machine is atypical to 99.99% recurring out there :-)

danilius, 2016-04-18:
Will give the ppa a whirl when I have a moment.

thopiekar, 2016-04-18:
@danilius By the way, it doesn't seem that you are alone with that problem :wink:

danilius, 2016-04-18:
Misery loves company.....

kakaroto, 2016-04-18:
After investigating the issue, we realized that the system provided sip.so gets loaded instead of the one bundled with cura. It turns out that the sys.path contains the path to the system's site-packages before the PYTHONPATH which is apparently caused by having a setuptools.pth file (more info here : https://stackoverflow.com/questions/1001851/pythonpath-ignored).
Anyways, here's how to fix it : 

* Edit /opt/cura/bin/cura
* Go to the 8th line (just after the `import os`)
* Add this line : `sys.path.insert(1, os.environ.get('PYTHONPATH', ''))`

This will re-add the PYTHONPATH before the system libraries, allowing the cura-provided sip.so to be loaded and fixing the problem.

danilius, 2016-04-18:
That worked a treat! Thanks!

Now, I'm not moaning, just asking/observing. The menu font is in serif, not the same as the rest of the UI. Is this my machine being 'orrible, or has anyone else observed that and is there anyone to force the menu to behave?

Also, would you like a dump of all the errors the command line spits out?

awhiemstra, 2016-04-19:
We probably know most of the errors, there are a bunch of them caused by bugs in Qt. But feel free to post it anyway, we might find something new. The font issue is known and is due to Qt not reading fonts from the system path (and then falling back to an ugly default serif font).

awhiemstra, 2016-04-19:
@kakaroto Nice that you found out what is causing it. I will see if I can add that to the main Cura file so that we do not run into this issue anymore.

Ghostkeeper, 2016-04-19:
@danilius The application menu font is (or rather, should be) determined by your system at the moment. It's a bit out of our reach since it is made natively. If we were to fix that, we'd probably have to implement a custom drop-down menu.

If it is not your default system font for application menus, I suggest you make a different issue about it.

danilius, 2016-04-19:
@Ghostkeeper, thanks for your reply. My system font is nothing like the one used in the menu, and I have no idea where it's coming from.

The menu font does not really bother me all that much, and if it's not a quickie fix I don't think it's worth raising an issue at this stage, more the sort of thing you can reserve for polishing.

closed 2016-04-28 by awhiemstra