Ultimaker/Cura #707: Auto slicing can't seem to start.

Gareee, 2016-04-18:
[image]

I think I kilt it. Each of these game tokens are about 1 1/4" across. I had no problem slicing 18 on old cura versions, but 2.1 chokes on these. It could be that auto slice got hung up while I was replicating and arranging them.

This is why I miss the replicate multiple option.

thopiekar, 2016-04-18:
The printing areas are overlapping and it also seems that the tokens on the left are placed out of the buildplate. I don't know why there are these gray areas, but they are wasting your build space a lot.

sedwards2009, 2016-04-18:
This sounds exactly like #594, and guessing by the window decorations it looks like you're running on Windows 10 too.

Gareee, 2016-04-18:
Yep. Windows 10. Why i have those huge gray areas is also beyond me.. I can understand not wanting to bump another object, but 2x the width on both sides? I'd think the gray surrounding would be square, since the object fits easily into a square. I'll check out that other issue report as well.

Ppyromann, 2016-04-18:
Try Special modes>print sequence>all at once
It looks like you have it set for 'one at a time'.  Those grey areas would be the clearance for the fan-duct.

Gareee, 2016-04-18:
The only problem with that is with the filament I'm using I get all sorts of stringing between the parts, and no amount of retraction changes of heat reduction get rid of it. Printing each one separately however does.

This is one of those instances where older versions of Cura are vastly superior than the new version.

I can replicate 17 copies with one click, they are auto positioned on the bed, and it can slice these to each print separately without a problem. Each of these are major features we've lost in the update.

nallath, 2016-04-19:
One at a time should be able to handle this. If not, it should show messages to the user. 
Could you give me the logs? I'm guessing this is the same issue as the other slicing won't start.

Ghostkeeper, 2016-04-19:
If it were caused by the one-at-a-time ordering algorithm, it wouldn't even get to "Preparing to slice...". Definitely a duplicate of #594.

closed 2017-02-24 by Ghostkeeper