Ultimaker/Cura #712: Request for FlashForge Creator Pro Machine Preset

hyperdo, 2016-04-20:
Hi All,

Would it be possible to get a FlashForge Creator Pro preset for Cura? In the menu it said to report it in the issue tracker, so here I am. :-)

Alternatively, if someone could provide me with another way to get Cura configured for the printer, I would be very grateful. I've used the software with other printers and have found it to be excellent. Thanks!

nallath, 2016-04-20:
Yes it's possible; if someone from the comunity provides us with a pull request. In that case i'd be more than happy to merge it so it will be in the next release (2.2) 

The manual provided by Ultimaker about cura has a bit of information about how to add new machines (and how to create them).

closed 2016-04-20 by nallath

markwal, 2016-04-20:
I have made one of these.  However, I've been reluctant to PR it into Cura for a couple of reasons.  First, the printer is a dual nozzle printer and I currently use three different machine definitions due to differences in the start.gcode for each scenario: 1. ffcpright.json for printing with the right nozzle only, 2. ffcpleft.json for printing with the left nozzle only, 3. ffcpdual.json for printing with both nozzles.

To be able to consolidate down to just one machine definition, but still allow the same flexibility, it'll take a PR or two to change a few more things with regard to start gcode variable expansion.  In particular, for "all objects are on the left extruder and all support/raft are also on the left extruder, switch to the left before any preheat and purge/wipe and only preheat one nozzle", it's a bit tricky since Cura's internals assume start on T0 and wants to retract it when switching to T1 even if it is stone cold and never going to print anything this print.

We could abandon the "print only with left nozzle" scenario, which would minimize the tweaks necessary, but I think a minimal necessary is to distinguish between single nozzle print and dual nozzle print.  That works in the case that no control over when the preheat occurs in the start gcode (let Cura emit the M104's, but for smooth printing on a FlashForge using some hosts it is necessary to get a M136 (start build) to be the first gcode, which then requires the preheat and wait for heat to come after that.  My first attempt at this was to have an variable expansion that was for all the heat commands, but that got nixed on my PR.

The other reason that makes me reluctant is that in spite of the fact that I own one as my first introduction to 3D printing at home, I'm now uncomfortably aware of how much FlashForge relies on the community to provide software and support for their machines and how little they're willing to give back.  Such that if we do include it here, it'll inevitably be a source of more folks in need of support using Cura with their FlashForge and it'll be up to the community to provide it (because FlashForge won't).  But on the other hand, perhaps I should give back to the community by providing the profile and help I suppose.

hongkongkiwi, 2016-05-01:
I'm very interested in this. I have a QiDI Tech printer which is a copy of FlashForge (but a bit cheaper).

I'm using Cura with the MakerBot profile, but it makes much more sense to have a dedicated profile for atleast Flashforge, or even better QiDi.