Ultimaker/Cura #714: ImportError: dynamic module does not define init function (PyInit_Arcus)

yushuiqiang, 2016-04-21:
jackyyu@jackyyu-All-Series:~/dev/Cura$ python3 cura_app.py
Traceback (most recent call last):
  File "cura_app.py", line 18, in <module>
    import Arcus #@UnusedImport
ImportError: dynamic module does not define init function (PyInit_Arcus)

sedwards2009, 2016-04-21:
Can you tell us what operating system you are using, and also which exact version of Cura you are using?

yushuiqiang, 2016-04-21:
OS is Ubuntu 15.04,Cura version is the cura-master

sedwards2009, 2016-04-21:
64 or 32 bit version?

yushuiqiang, 2016-04-21:
64  bit version?

Ghostkeeper, 2016-04-21:
I might hope it's 64-bit. 32-bit Ubuntu is rare nowadays and unsupported by us!

How did you compile Arcus, I wonder? We are only compiling Arcus with SIP, not distutils.

yushuiqiang, 2016-04-21:
I use this sh file;

cd ~
if [ ! -d "dev" ]; then
    mkdir dev
fi
cd dev
 
sudo apt-get install -y git cmake cmake-gui autoconf libtool python3-setuptools curl python3-pyqt5.* python3-numpy qml-module-qtquick-controls
 
#protobuf.
#https://github.com/google/protobuf.git
 
echo "================================="
echo "Install Protobuf."
if [ ! -d "protobuf" ]; then
    git clone https://github.com/Ultimaker/protobuf.git
    cd protobuf
else
    cd protobuf
    git pull        
fi
echo "================================="
echo "get gtest."
if [ ! -d "gtest" ]; then
    git clone https://github.com/kgcd/gtest.git
else
    git pull
fi
echo "================================="
echo "get gmock."
if [ ! -d "gmock" ]; then
    git clone https://github.com/krzysztof-jusiak/gmock.git
else
    git pull        
fi
echo "Build Protobuf."
./autogen.sh
./configure --prefix=/usr
make -j4
sudo make install
sudo ldconfig
cd python
python3 setup.py build
sudo python3 setup.py install
cd ../..
 
echo "================================="
echo "Install libArcus."
if [ ! -d "libArcus" ]; then
    git clone https://github.com/Ultimaker/libArcus
    cd libArcus
else
    cd libArcus
    git pull        
fi
if [ ! -d "build" ]; then
  mkdir build
fi
cd build
#cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DPYTHON_SITE_PACKAGES_DIR=/usr/lib/python3.4/dist-packages
cmake .. -DPYTHON_SITE_PACKAGES_DIR=/usr/lib/python3.4/dist-packages
make -j4
sudo make install
cd ../../
 
echo "================================="
echo "Install CuraEngine."
if [ ! -d "CuraEngine" ]; then
    git clone https://github.com/Ultimaker/CuraEngine.git
    cd CuraEngine
else
    cd CuraEngine
    git pull        
fi
if [ ! -d "build" ]; then
  mkdir build
fi
cd build
#cmake .. -DCMAKE_INSTALL_PREFIX=/usr
cmake .. 
make -j4
sudo make install
cd ../../
 
echo "================================="
echo "Install Uranium."
if [ ! -d "Uranium" ]; then
    git clone https://github.com/Ultimaker/Uranium.git
    cd Uranium
else
    cd Uranium
    git pull        
fi
if [ ! -d "build" ]; then
  mkdir build
fi
cd build
#cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DPYTHON_SITE_PACKAGES_DIR=/usr/lib/python3.4/dist-packages -DURANIUM_PLUGINS_DIR=/usr/lib/python3.4/dist-packages
cmake .. -DPYTHON_SITE_PACKAGES_DIR=/usr/lib/python3.4/dist-packages -DURANIUM_PLUGINS_DIR=/usr/lib/python3.4/dist-packages
sudo make install
cd ../..
 
echo "================================="
echo "Install Cura."
if [ ! -d "Cura" ]; then
    git clone https://github.com/Ultimaker/Cura.git
    cd Cura
else
    cd Cura
    git pull        
fi
cd ..
echo "Build finished."
 
echo "============================================================================"
echo "Merge Resource into Cura/resources/"
cp -rv Uranium/resources/* Cura/resources/
echo "Merge Plugins into Cura/plugins/"
cp -rv Uranium/plugins/* Cura/plugins/
echo "Link:"$PWD"/CuraEngine/build/CuraEngine"
sudo ln -s $PWD/CuraEngine/build/CuraEngine /usr/bin/CuraEngine
 
echo "Starting Cura......"
cd Cura
python3 cura_app.py
echo "You need add to /etc/profile:export PYTHONPATH=/usr/lib/python3/dist-packages"
echo "============================================================================="

Ghostkeeper, 2016-04-21:
Huh, should work all right then. Looks okay at first glance.

fieldOfView, 2016-04-23:
A workaround is to comment out line 18 (the unused Arcus import)

990092230, 2016-08-05:
Hi, I have this issue , too. Have you fix it ?

Ghostkeeper, 2016-08-05:
We haven't changed anything there as far as I know.

closed 2017-07-27 by nallath