Ultimaker/Cura #715: Custom Printer Profiles

JeffLatham, 2016-04-22:
The user is only able to choose from a short list of predefined printers. The ability to create a custom printer profile is desperately needed (ability to define print volume and other parameters).

The-Experimentalist, 2016-04-24:
Same problem here, would love to have a look but I can't due to this issue.

fieldOfView, 2016-04-25:
There is a way to create a custom printer profile, it is just not (yet) integrated into Cura. Please read the note at the end of the readme:
https://github.com/Ultimaker/Cura/blob/master/README.md

nallath, 2016-04-25:
https://ultimaker.com/en/resources/20511-change-machine-settings

closed 2016-04-25 by nallath

JeffLatham, 2016-04-25:
Followed these instructions. Created a custom profile (.json file) for my
printer. It's still not available as an option within Cura when I go to add
a printer...

fieldOfView, 2016-04-25:
Could you post the file you created somewhere? (eg http://gist.github.com)

JeffLatham, 2016-04-25:
Generated using this tool: http://quillford.github.io/CuraProfileMaker/
Named: Printrbot Simple Metal.json
Located: Contents>Resources>cura>resources>machines
File Contents:

{
    "id": "Printrbot Sime Metal",
    "name": "Printrbot",
    "icon": "icon_ultimaker.png",
    "platform": "",
    "inherits": "fdmprinter.json",
    "machine_settings": {
        "machine_width": {
            "default": 250
        },
        "machine_height": {
            "default": 150
        },
        "machine_depth": {
            "default": 150
        },
        "machine_center_is_zero": {
            "default": false
        },
        "machine_nozzle_size": {
            "default": 0.4
        },
        "machine_head_shape_min_x": {
            "default": 75
        },
        "machine_head_shape_min_y": {
            "default": 18
        },
        "machine_head_shape_max_x": {
            "default": 18
        },
        "machine_head_shape_max_y": {
            "default": 35
        },
        "machine_nozzle_gantry_distance": {
            "default": 55
        },
        "machine_nozzle_offset_x_1": {
            "default": 18
        },
        "machine_nozzle_offset_y_1": {
            "default": 0
        },
        "machine_gcode_flavor": {
            "default": "RepRap (Marlin/Sprinter)"
        },
        "machine_start_gcode": {
            "default": ";Sliced at: {day} {date} {time}\n;Basic
settings: Layer height: {layer_height} Walls: {wall_thickness} Fill:
{fill_density}\n;Print time: {print_time}\n;Filament used:
{filament_amount}m {filament_weight}g\n;Filament cost:
{filament_cost}\n; From Slic3r\nG21          ; set units to
millimeters\nG90          ; absolute positioning\nM82          ; set
extruder to absolute mode\nG92 E0       ; zero the extruded length\n;
Start GCode\nM107         ; fan off\nM109         ; wait for extruder
temperature\nM190         ; wait for bed temperature\nG28 X0 Y0 Z0 ;
home X & Y & Z axes\nM211 X230    ; reduce X max\nM500         ; save
settings\nG29          ; bed leveling sequence\nM211 X250    ; restore
X max\nM500         ; save settings"
        },
        "machine_end_gcode": {
            "default": "; End GCode\nM104 S0       ; turn off extruder
temperature\nM140 S0       ; turn off bed temperature\nM107          ;
fan off\nG28 X0 Y0     ; home X & Y axes\nM84           ; motors off"
        }
    },
    "categories": {
        "material": {
            "settings": {
                "material_bed_temperature": {
                    "visible": true
                }
            }
        }
    }
}

fieldOfView, 2016-04-25:
The (biggest) problem is that there was no "version" key in the file. Here's a fixed version:
https://gist.github.com/fieldOfView/de814833cb663398460e0d3944c1fbcd
Note that I have not checked any of the settings to work with your printrbot, I have just checked why it wasn't showing up in Cura.

JeffLatham, 2016-04-25:
Works! Thanks!

Ghostkeeper, 2016-04-26:
If it works, we could consider adding a Printrbot definition to Cura?

fieldOfView, 2016-04-26:
The start gcode would need some work. It contains a number of {keywords} that won't work (though from a cursory view it looks like they are in comments, so non-critical)

On a sidenote: @quillford's CuraProfileMaker has been updated to include a version in the generated machine definition.

JeffLatham, 2016-04-26:
My gcode that I put in the file is for a super specific custom setup that I have. Best to not use it unless you happen to have a Printrbot Simple Metal with RevD Printrboard and an upgraded X axis...