Ultimaker/Cura #716: ImportError: No module named 'Arcus'

PauloMFerreira, 2016-04-22:
Hi!

Im trying to debugging Cura and I got this error:
File "/Users/PauloFerreira/Cura-git/Cura/cura_app.py", line 7, in <module>
    import Arcus
ImportError: No module named 'Arcus'

Im already install Protobuf and build libArcus.
Im using MacOSX.

Thanks.

Ghostkeeper, 2016-04-24:
Your Python installation can't find libArcus. Your build of libArcus should have produced a file called Arcus.so. You need to set your `PYTHONPATH` environment variable to include the directory where that .so file is located.

I don't know by heart where OSX puts the python libraries typically.

PauloMFerreira, 2016-04-27:
Thanks, that worked fine.

But now Im getting another error.

```
2016-04-27 12:06:32,212 - ERROR - Unable to find backend executable: /Users/PauloFerreira/Cura-git/bin/CuraEngine
2016-04-27 12:06:32,239 - DEBUG - Setting active view to SolidView
QQmlApplicationEngine failed to load component
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:381 Type UM.MessageStack unavailable
file:///Users/PauloFerreira/Cura-git/Cura/UM/Qt/qml/UM/MessageStack.qml:6 module "QtQuick.Controls.Styles" version 1.3 is not installed
```

Its seems that is trying to load CuraEngine at cura/bin folder but it doesnt exist. CuraEngine is in plugins folder. 
And then it required QtQuick.Controls.Styles 1.3 and I already install  Qt5.3.2. That supposed installed, right?

fieldOfView, 2016-04-27:
You need to install qml-module-qtquick-controls and qml-module-qtquick-layouts, which don't come with Qt automatically.

The CuraEngingBackend in the plugins folder only contains the logic to communicate with the CuraEngine executable, not the CuraEngine itself. The binary executable for CuraEngine is supposed to be in that bin folder. You can create a link to the version in the build directory of CuraEngine (supposing you have built CuraEngine from source)

PauloMFerreira, 2016-04-27:
Right. CuraEngine is loading now.
But I dont know how to install qml-module-qtquick-controls and qml-module-qtquick-layouts on MacOSX. In Linux, it seems to be quite simple with apt-get. Any suggestions to install it on MacOSX?

fieldOfView, 2016-04-27:
I think you have to update to Qt5.4 or newer.

PauloMFerreira, 2016-04-27:
Already install Qt5.6 now. And give me a similar error:

```
QQmlApplicationEngine failed to load component
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:5 module "QtQuick.Controls" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:6 module "QtQuick.Controls.Styles" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:4 module "QtQuick" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:7 module "QtQuick.Layouts" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:8 module "QtQuick.Dialogs" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:5 module "QtQuick.Controls" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:6 module "QtQuick.Controls.Styles" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:4 module "QtQuick" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:7 module "QtQuick.Layouts" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:8 module "QtQuick.Dialogs" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:5 module "QtQuick.Controls" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:6 module "QtQuick.Controls.Styles" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:4 module "QtQuick" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:7 module "QtQuick.Layouts" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:8 module "QtQuick.Dialogs" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:5 module "QtQuick.Controls" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:6 module "QtQuick.Controls.Styles" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:4 module "QtQuick" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:7 module "QtQuick.Layouts" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:8 module "QtQuick.Dialogs" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:5 module "QtQuick.Controls" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:6 module "QtQuick.Controls.Styles" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:4 module "QtQuick" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:7 module "QtQuick.Layouts" is not installed
file:///Users/PauloFerreira/Cura-git/Cura/resources/qml/Cura.qml:8 module "QtQuick.Dialogs" is not installed
```

Its strange because in Qt folder (Qt/5.6/clang_64/qml) theres a folder "QtQuick" and inside have Controls, Dialogs and Layouts folder.

awhiemstra, 2016-04-28:
It sounds like it cannot find the QtQuick plugins. Try setting the QML2_IMPORT_PATH environment variable to that directory.

PauloMFerreira, 2016-04-28:
I create that environment variable to "~/Qt/5.6/clang_64/bin". When I run printenv on terminal it is there. But the error is the same. I search on the internet and I dont know what Im doing wrong.

PauloMFerreira, 2016-04-28:
My mistake. I put the variable to "/Qt/5.6/clang_64/qml" and its work now.

Thanks

closed 2016-04-29 by Ghostkeeper