Ultimaker/Cura #717: Build volume contour shown incorrectly

thopiekar, 2016-04-22:
It is hard to explain so I made a video. I noticed that in ```2.1``` and ```master```.

https://www.youtube.com/watch?v=hL4XaFJb1vY

thopiekar, 2016-04-22:
Btw. Ultimaker2 is shown correctly..

fieldOfView, 2016-04-23:
Weird. Is there anything special you need to do to reproduce, apart from selecting the BQ Hephaistos 2? Sofar I have been unable to reproduce this on windows.

thopiekar, 2016-04-23:
I don't think there will be anything special in the config file. I can
restart my computer to Windows 10 and retry it there.

thopiekar, 2016-04-23:
Doesn't happen in Cura 2.1 on Windows 10 installed via installer from
the Ultimaker homepage.
Is there any info which codebase was used  to build the Cura bundle?
Eg. the last commit in Cura/Uranium/libArcus/etc? So I can rebuild my
packages in my PPA on this older codebase to see whether it is related
to a dependency (newer Qt5/PyQt/Protobuf version?) or not.

2016-04-23 11:34 GMT+02:00 Thomas Pietrowski <thopiekar@gmail.com>:
> I don't think there will be anything special in the config file. I can
> restart my computer to Windows 10 and retry it there.

thopiekar, 2016-04-23:
I synced the repositories at launchpad with GitHub and rebuilt my packages. Now it is working again :grin: 
Seems that someone was working on something when I made the sync the last time, so something was missing...

closed 2016-04-23 by thopiekar